CopterExpress/clover #19: Acceleration build with APT-FAST

urpylka, 2018-02-27:
An essential part of build time consist installation packages with apt (apt-get) in own thread. What do you think about using **apt-fast** utility? https://github.com/ilikenwf/apt-fast
```(bash)
sudo alias apt-get='apt-fast'
```

korigod, 2018-02-27:
`apt-fast` uses `axel` to download packages from multiple mirrors. We can just set up a local debian mirror instead, it's just several hundreds of gigabytes for all packages (https://www.debian.org/mirror/size). This way we will not depend on our internet connection at all.

urpylka, 2018-04-23:
Ok, we made decision, that we don't use apt-fast.
---
Also we made local mirror with debian packages. Our scripts avaible in our [private repo](https://github.com/CopterExpress/coex-mirror).

closed 2018-04-23 by urpylka