CopterExpress/clover #38: Use raspi-config for configure hardware interfaces

urpylka, 2018-04-23:
I propose used for configured hardware interfaces `raspi-config` f.e. `/usr/bin/raspi-config nonint do_i2c 0`. Now we used forked scripts from `raspi-config`.

urpylka, 2018-05-04:
I replaced our forked-scripts to ‘raspi-config’ https://github.com/CopterExpress/clever/commit/0346c48546590faa00f7ee0a1d0ea71f6343bd17

But I meet problem with lock version of raspi-config.
F.e. I know code (to enable only hardware uart, wo console) that I wanted to use, added to ‘raspi-config’ only 4 days ago.

I propose several solutions:
1. Add ‘raspi-config’ as submodule to repository
2. Add execute file of ‘raspi-config’ to this repository as file
3. Replace ‘raspi-config’ in repo to fixed version

urpylka, 2018-05-04:
Also I found bug in log of `image_builder`. So in `chroot` doesn't work `dtparam` util.

andrew-xdn, 2018-05-04:
Your "config" directory has a really strange location. I thing the right place for it is in "/sys/kernel", not in root. Are you sure you run "dtparam" in the correct environment? Still, I think it's not ok to run utility that use kernel or hardware resources under chroot. Isn't it better to edit "/boot/config.txt" directly in this case?

urpylka, 2018-05-05:
@andrew-xdn I checked environment, it's normal.
And `raspi-config` change `/boot/config.txt`, but also trying use dynamic modules. I don't mind that normal run kernel-utility in chroot. 

Because temporarily I made hack in PR https://github.com/CopterExpress/clever/pull/42. Let's discuss!

urpylka, 2018-05-05:
I found solution how we can enable HW UART interface wo user console with current version of `raspi-config` in non-interactive mode. More there https://github.com/CopterExpress/clever/pull/43.

Now we haven't problems with lock version of `raspi-config`.

urpylka, 2018-10-10:
Now we use RPi init script w raspi-config https://github.com/CopterExpress/clever/pull/68

closed 2018-10-10 by urpylka