CopterExpress/clover #50: Remove build_isolated from the image, if it's not needed

okalachev, 2018-06-01:


urpylka, 2018-06-01:
I made that in my local branch. After build image & tests I make decision about make this in our `master` branch.

urpylka, 2018-06-02:
Test result:

```bash
2.3G	/mnt/hdd_builder/workspace/clever_remove_build_dirs_01062018.img.zip
2.4G	/mnt/hdd_builder/workspace/clever_v0.6_2018-05-10T13:41:27Z.img.zip
```

Maybe, if we will shrink free space on image this may be helpful for upload image to Ya.Disk.

Stragenly, I got error:
```
-bash: /home/pi/catkin_ws/devel/setup.bash: No such file or directory
```
But clever.service & rostopic it's work.

@goldarte make full test this image in Monday .

urpylka, 2018-06-06:
@goldarte As main option you need to test possibility used Clever ROS API in another nodes, after remove `catkin_ws/devel`. And also you need to test how work Clever node.

urpylka, 2018-06-21:
Add commits
https://github.com/CopterExpress/clever/commit/1e2fd40c05e118f4ebccfe8ff64443652f128424
https://github.com/CopterExpress/clever/commit/06a01f7e320802628a8af753c022e5d2a4620957

closed 2018-06-21 by urpylka