CopterExpress/clover #60: Запуск Клевера с разрешением 640x480, камера в потолок

copterspace, 2018-08-08:
Попробовал запустить Клевер на 640х480 (вместо 320х240 по умолчанию). На потолочных маркерах.
В main_camera.launch прописал:
* param name="camera_info_url" value="file://$(find clever)/camera_info/fe130_640_02.yaml"
* param name="image_width" value="640"
* param name="image_height" value="480"

_fe130_640_02.yaml - калибровал как описано в http://wiki.ros.org/camera_calibration , потом полученные коэффициенты вписал в образцовый fisheye_cam_640.yaml_  

С помощью rviz проверил на дизармленом коптере - карту на потолке видит, маркеры распознаёт, положение в пространстве и движение по всем трём осям видит правильно. 

Но при взлёте на 1 метр и зависании (с помощью navigate) - коптер взлетает, плавно пролетает target position и летит вверх, пока не ударится в потолок. В rviz это видно также как и в логах:
[image]

LPE_VIS_XY = 0.1 m, LPE_VIS_Z = 0.15 m и остальные настройки pixhawk- те же, что и при 320х240.
При полётах с разрешением камеры по умолчанию 320х240 такой проблемы не наблюдается, коптер зависает в target position, a не пролетает её.

В чём может быть проблема и куда смотреть?

okalachev, 2018-08-08:
А на той калибровке, что уже лежит в Клевере под 640х480, не пробовал?

copterspace, 2018-08-09:
А подлёт с разрешением 320х240 выглядит вот так (только разрешение камеры поменял, остальное без изменений):
[image]
 
Коптер видит, что по Z он взлетел на 1,5 м, а не на 1м. Но дальше он в потолок не бьётся - только мотается выше целевой точки, и через 30 секунд по команде land() садится. pos_mocap.z и local_position.z не сильно расходятся на всём полёте.

okalachev, 2018-08-10:
Вообще говоря, есть планы попробовать завести всю эту систему на EKF2. Возможно, какие-то проблемы уйдут.

В любом случае, я изучу твои логи. Хотя желательно, все-таки, приложить их в виде файлов.

copterspace, 2018-08-10:
Да, конечно, логи высылаю.
[logs.zip](https://github.com/CopterExpress/clever/files/2278117/logs.zip)
3, 4 - попытки взлёта 640х480 с типовой калибровкой камеры, неудачные
5 - когда вернул в 320х240, удачный

Вот ещё лог 2 - взлёт 640х480 с нетиповой калибровкой, полученной с помощью camera_calibrator.
[log_2.zip](https://github.com/CopterExpress/clever/files/2278136/log_2.zip)

korigod, 2019-04-18:
@copterspace, issue ещё актуален? Как обстоит дело на последней версии образа?

copterspace, 2019-04-19:
На следующей неделе могу поставить камеру наверх, скачать последний образ и протестить. С тех пор мы летали только на 320х240.

copterspace, 2019-04-23:
Чё-то новый образ (брал от НТИ, clever_0.16-nti2019-nvg.1.img ) не хочет есть нашу карту.
Сгенерировал карту скриптом rosrun aruco_pose genmap.py. С виду всё хорошо (файл прилагаю [cospa_map.txt](https://github.com/CopterExpress/clever/files/3107686/cospa_map.txt)).
Но после того как я пытаюсь подставить этот файл в aruco.launch вместо nti_novgorod.txt - при запуске Клевер падает с ошибкой:

OpenCV Error: Assertion failed (id < bytesList.rows) in drawMarker, file /home/pi/ros_catkin_ws/src/opencv3/opencv_contrib/aruco/src/dictionary.cpp, line 167
[FATAL] [1542130560.742547325]: Failed to load nodelet '/aruco_detect` of type `aruco_pose/aruco_detect` to manager `nodelet_manager'
[FATAL] [1542130560.742933992]: Failed to load nodelet '/aruco_map` of type `aruco_pose/aruco_map` to manager `nodelet_manager'
[FATAL] [1542130560.743252273]: Failed to load nodelet '/main_camera` of type `cv_camera/CvCameraNodelet` to manager `nodelet_manager'
[FATAL] [1542130560.743684252]: Failed to load nodelet '/optical_flow` of type `clever/optical_flow` to manager `nodelet_manager'
[nodelet_manager-9] process has died [pid 19111, exit code -11, cmd /opt/ros/kinetic/lib/nodelet/nodelet manager __name:=nodelet_manager __log:=/home/pi/.ros/log/f59908de-e767-11e8-a266-b827eb346319/nodelet_manager-9.log].
log file: /home/pi/.ros/log/f59908de-e767-11e8-a266-b827eb346319/nodelet_manager-9*.log
[mavros-1] process has finished cleanly
log file: /home/pi/.ros/log/f59908de-e767-11e8-a266-b827eb346319/mavros-1*.log
[aruco_detect-5] process has died [pid 19060, exit code 255, cmd /opt/ros/kinetic/lib/nodelet/nodelet load aruco_pose/aruco_detect nodelet_manager image_raw:=main_camera/image_raw camera_info:=main_camera/camera_info __name:=aruco_detect __log:=/home/pi/.ros/log/f59908de-e767-11e8-a266-b827eb346319/aruco_detect-5.log].
log file: /home/pi/.ros/log/f59908de-e767-11e8-a266-b827eb346319/aruco_detect-5*.log
[optical_flow-8] process has died [pid 19102, exit code 255, cmd /opt/ros/kinetic/lib/nodelet/nodelet load clever/optical_flow nodelet_manager image_raw:=main_camera/image_raw camera_info:=main_camera/camera_info __name:=optical_flow __log:=/home/pi/.ros/log/f59908de-e767-11e8-a266-b827eb346319/optical_flow-8.log].
log file: /home/pi/.ros/log/f59908de-e767-11e8-a266-b827eb346319/optical_flow-8*.log
[main_camera-14] process has died [pid 19147, exit code 255, cmd /opt/ros/kinetic/lib/nodelet/nodelet load cv_camera/CvCameraNodelet nodelet_manager __name:=main_camera __log:=/home/pi/.ros/log/f59908de-e767-11e8-a266-b827eb346319/main_camera-14.log].
log file: /home/pi/.ros/log/f59908de-e767-11e8-a266-b827eb346319/main_camera-14*.log
[aruco_map-6] process has died [pid 19076, exit code 255, cmd /opt/ros/kinetic/lib/nodelet/nodelet load aruco_pose/aruco_map nodelet_manager image_raw:=main_camera/image_raw camera_info:=main_camera/camera_info markers:=aruco_detect/markers __name:=aruco_map __log:=/home/pi/.ros/log/f59908de-e767-11e8-a266-b827eb346319/aruco_map-6.log].
log file: /home/pi/.ros/log/f59908de-e767-11e8-a266-b827eb346319/aruco_map-6*.log

Ставлю обратно nti_novgorod.txt - ошибка исчезает.
Т.е. понятно, что где-то в карте собака порылась - но её ж такую скрипт сгенерил, и при осмотре текстового файла внешних изъянов я пока не нашёл...

okalachev, 2019-04-23:
У нас по умолчанию в `aruco_detect` выбран словарь на 250 маркеров (DICT_4X4_250).

Чтобы поменять словарь на DICT_4X4_1000, надо добавить параметр `dictionary` в настройки `aruco_detect`'а (https://github.com/CopterExpress/clever/blob/master/clever/launch/aruco.launch#L9):
```xml
<param name="dictionary" value="3"/>
```

[Вот](https://github.com/CopterExpress/clever/tree/master/aruco_pose#parameters), кстати, список словарей.

copterspace, 2019-04-24:
Спасибо, Олег. Не только `aruco_detect`'a, но и ещё и `aruco_map`'a, если быть точным - там он также пытается взять словарь из параметра ноды (стр.91), и именно из-за `aruco_map`'a он и падал, как оказалось.
Ок, карта отобразилась, поеду дальше...

copterspace, 2019-04-24:
На 640х480 получается маловатая частота расчёта позы карты, имхо:
```
pi@raspberrypi:~ $ rostopic hz /aruco_map/pose
subscribed to [/aruco_map/pose]
average rate: 10.478
        min: 0.056s max: 0.176s std dev: 0.04461s window: 10
average rate: 10.794
        min: 0.054s max: 0.176s std dev: 0.04128s window: 21
average rate: 10.568
        min: 0.054s max: 0.176s std dev: 0.04029s window: 31
average rate: 10.643
        min: 0.054s max: 0.176s std dev: 0.03784s window: 42
average rate: 10.365
        min: 0.054s max: 0.176s std dev: 0.03687s window: 51

```
И это на 1 маркере. Если карту показать - ещё на %10-20 падает framerate.
На 320х240 Клевер 16 выдаёт мне 20-22 Гц максимум. А 13-й Клевер у меня в тех же условиях выдаёт 33-35 Гц. 
С 10-герцовой частотой позы я на своём одометре пробовал летать - было сильно "не очень".
Полётные тесты позже проведу, но из-за низкой частоты прогноз на летабельность 640х480 пока отрицательный... 
У вас получается ли больший framerate для `rostopic hz /aruco_map/pose` из RPI выжать? (может, у меня чего не так настроено...)

okalachev, 2019-04-24:
Т. е. ты говоришь, что с 13-й версии до 16-й образа частота распознавалки арук деградировала? Мы просто этого не замечали.
@Alamoris

copterspace, 2019-04-30:
Сегодняшние полётные тесты подтвердили предположения о деградации частоты распознавания арук, пока только в POSCTL.
320х240 - висит в точке хоть как-то, 640х480 (поменялось **только** разрешение камеры) - не держит точку совсем. Отличается только framerate, в rviz карту видно примерно одинаково в обоих случаях. Кстати можно ещё данный тезис проверить поставив framerate = 10 для 320х240 - должна тоже улетать прочь...
примеры логов:
320х240 - https://logs.px4.io/plot_app?log=d8068dab-702a-4382-bfeb-e557e6e4a901
640х480 - https://logs.px4.io/plot_app?log=60d47fed-5f02-4d82-a645-669b4f660886

для сравнения, на 13-м Клевере летала более стабильно, примерно вот так: https://www.youtube.com/watch?v=RNG2Y_IHx6Y

okalachev, 2019-04-30:
А калибровку ты меняешь?

copterspace, 2019-04-30:
Камеры? - Да. Я пробовал и с нашей и с не нашей. С нашей калибровкой всё лучше, до потолка 2.4м, а с вашей она думает, что до карты на потолке 1.4 м.

okalachev, 2019-04-30:
Калибровка камера должна соответствовать разрешению. На каждое разрешение своя калибровка. (Хотя можно и автоматически camera_matrix пересчитывает, `cv_camera` это, кстати умеет).

copterspace, 2019-05-06:
Калибровка камеры своя для каждого разрешения, конечно.
Сегодня отключил optical flow, дальномер в clever.launch и estimate_poses в aruco_detect.
Производительности стало лучше, на 320х240 framerate вернулся практически к 40.
После переключения в 640х480 - из `/aruco_map/pose` максимум удалось выжать где-то 12 герц. 
На 12 герцах в POSCTL, в принципе, аппарат висит. Но быстро летать - не очень получится...
При этом loadavg на 320х240 = ~4.7, а на 640х480 - уже аж 5.9 - 6.3. Т.е., похоже, упираемся в физический предел производительности.
Думаю, на этом issue можно закрывать. Спасибо за консультации!

okalachev, 2019-05-06:
@Alamoris, а у нас вроде лучше результаты? Вполне возможно, что на разных Raspberry могут быть разные результаты.

closed 2019-05-06 by okalachev