CopterExpress/clover #61: Пока просто заметка

urpylka, 2018-08-23:
Наш программный пакет состоит из четырех основных частей:
1. ROS пакет CLEVER, в свою очередь содержит:
    1. Нода для определения положения аппарата
    2. Нода API CLEVER (Simple Control)
2. Набор инструкций для сборки образа операционной системы
3. Документация к ROS, аппаратной сборки и настройки коптера
4. Исходники приложения программного пульта под iOS

Почему считаю что лучше переносить.

Основная идея заключается в том что
Концептуально Arucopose И simpleoffboard являются лишь устанавливаемыми пакетами
И нет причин держать их в главном пакете
А документация должна прикручиваться исходя из того что добавляется в install_ros или другие requirements как и эти репозитории.

Также использование их как внешних пакетов должно наконец стимулировать решить проблему с ros requirements

Вот пока не понял только где должна быть документация 😅
CLEVER - это набор инструкций для сборки преднастроенного образа с документацией.

1. Что связывает ноду клевера и builder?
2. Что связывает документацию и builder?

okalachev, 2018-08-23:
Так в пакете Clever помимо simple_offboard еще миллион нод.

urpylka, 2018-09-18:
[image]

okalachev, 2018-09-18:
simple_offoard и aruco_pose в будущем отдельные репы, сейчас пока не потяну.

smirart, 2018-09-18:
А что, если я сделаю так, чтобы при push или release в отдельную репу, пакет полностью собирался и добавлялся в нашу deb-репу? Мб даже из релизов можно будет скачать?

urpylka, 2018-09-18:
А что, если я сделаю так, чтобы при push или release в отдельную репу, пакет полностью собирался и добавлялся в нашу deb-репу? Мб даже из релизов можно будет скачать?

urpylka, 2018-10-10:
Обновил структуру CLEVER согласно этой заметке https://github.com/CopterExpress/clever/pull/68

closed 2018-10-10 by urpylka