CopterExpress/clover #69: Bug with import module rospy

xsestech, 2018-10-09:
When I started app I had this errors
  import rospy
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/__init__.py", line 47, in <module>
    from std_msgs.msg import Header
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/std_msgs/msg/__init__.py", line 1, in <module>
    from ._Bool import *
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/std_msgs/msg/_Bool.py", line 5, in <module>
    import genpy
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/genpy/__init__.py", line 34, in <module>
    from . message import Message, SerializationError, DeserializationError, MessageException, struct_I
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/genpy/message.py", line 44, in <module>
    import yaml
`# -*- coding: utf-8 -*-
import rospy
from clever import srv
from std_srvs.srv import Trigger
import rospy
from clever import srv
from std_srvs.srv import Trigger

rospy.init_node('xopter')   # название вашей ROS-ноды

# Создаем прокси ко всем сервисам:

get_telemetry = rospy.ServiceProxy('get_telemetry', srv.GetTelemetry)
navigate = rospy.ServiceProxy('navigate', srv.Navigate)
navigate_global = rospy.ServiceProxy('navigate_global', srv.NavigateGlobal)
set_position = rospy.ServiceProxy('set_position', srv.SetPosition)
set_velocity = rospy.ServiceProxy('set_velocity', srv.SetVelocity)
set_attitude = rospy.ServiceProxy('set_attitude', srv.SetAttitude)
set_rates = rospy.ServiceProxy('set_rates', srv.SetRates)
land = rospy.ServiceProxy('land', Trigger)
release = rospy.ServiceProxy('release', Trigger)
z = 1  # высота
tolerance = 0.2  # точность проверки высоты (м)

# Запоминаем изначальную точку
start = get_telemetry()

# Взлетаем на 2 м
print(navigate(z=z, speed=0.5, frame_id='fcu_horiz', auto_arm=True))

# Ожидаем взлета
while True:
    # Проверяем текущую высоту
    if get_telemetry().z - start.z + z < tolerance:
        # Взлет завершен
        break
    rospy.sleep(5)
res = land()
if res.success:
    print("Copter is landing")
`

urpylka, 2018-10-09:
What is your image version?

It’s like as error when you haven’t ros enviroment. Use: /opt/ros/kinetic/setup.bash

okalachev, 2018-10-09:
What is the image version?
@urpylka

xsestech, 2018-10-09:
[image]

xsestech, 2018-10-09:
> What is the image version?
> @urpylka

[image]

urpylka, 2018-10-09:
All clever-images (see releases on github) have a version. Use `cat /etc/clever_release` for see that

urpylka, 2018-10-09:
Or you use clever (ros package) without image? @xsestech

xsestech, 2018-10-09:
> /opt/ros/kinetic/setup.bash

I had error : 
 /opt/ros/kinetic/setup.bash: command not found

xsestech, 2018-10-09:
I am using image from official site.

xsestech, 2018-10-09:
> What is your image version?
> 
> It’s like as error when you haven’t ros enviroment. Use: /opt/ros/kinetic/setup.bash

I had an error:
sudo: /opt/ros/kinetic/setup.bash: command not found

urpylka, 2018-10-09:
I can not help you if you don’t give me more information.

xsestech, 2018-10-09:
I coppied image from  https://github.com/CopterExpress/clever/releases to cd card and connected by ssh to rasberry pi and wrote this  python script.
[image] and had this output
[image]

urpylka, 2018-10-09:
Error w command not found says that you haven’t ROS on your image. Please before use our image read documentation on gitbook. Or use our preconfigured **clever-image** (you can download it in releases on gh)

xsestech, 2018-10-09:
I downloaded clever-image from releases from github

urpylka, 2018-10-09:
That’s better. Next say me your image version. Execute `cat /etc/clever_version` on your image.

xsestech, 2018-10-09:
v0.11.4

urpylka, 2018-10-09:
Great 👍 The problem was caused because this image is not tested. Use image with `Latest release` label. For example v0.8.1

urpylka, 2018-10-09:
I guess that is new stable version will be created on next week

closed 2018-10-09 by urpylka

xsestech, 2018-10-09:
Ok