CopterExpress/clover #116: Сеть сломалась в 16-м Клевере

copterspace, 2019-04-26:
В 13-м Клевере и до него всё было хорошо: была у меня виртуалка Ubuntu с rviz-ом на ней, заполнял на виртуалке я 2 переменные: 
```
export ROS_MASTER_URI=http://192.168.11.1:11311
export ROS_IP=10.0.2.15

```
После этого у меня был полностью доступен ROS с Малины - и rostopic echo, и rviz и всё остальное. (10.0.2.15 - это виртуалкин ip адрес из ifconfig).

В 16-м Клевере сейчас:

- `rostopic list` - отображаются все топики, как на виртуалке, так и на Малине
- `rostopic echo /mavros/state` - есть только в сеансе малины и пусто - на виртуалке.
- попробовал через `rostopic pub` публиковать строку в тестовый топик на Малине - она также не отображается на виртуалке.

Тогда я остановил Клевер и начал лечить roscore, чтоб через  rostopic pub просто текст на виртуалке увидеть. И вылечился roscore только от возврата на место ROS_IP + устранения запуска сервиса roscore от пользователя pi ( отката https://github.com/CopterExpress/clever/pull/101  + отката https://github.com/CopterExpress/clever/commit/a89dda85764f5d4caa54216e298f15db2bd9513b#diff-e0fb8eab5334899c39004d5843582224 ) 

при этом у Клевера я User=pi пока не убрал из  builder/assets/clever.service - и rostopic echo /mavros/state по - прежнему виден ОК по ssh в Малине и НЕ виден на виртуалке. 

Я-то могу всё восстановить как было в 13-м Клевере, и наверно, заработает. Но вопрос: а как должно быть на самом деле?

p.s. на виртуалке пытался писать export ROS_HOSTNAME=raspberrypi.local - не спасло, пока пользователя pi в запуске сервисов не отключил - всё было глухо...

okalachev, 2019-04-26:
Спасибо за багрепорт!

Мы активно думаем над конфигурацией `ROS_IP`/`ROS_HOSTNAME` и пока что-то не можем придумать варианта, который хорошо работал бы во всех сетепах.

@sfalexrog

copterspace, 2019-04-26:
а ROS_IP старый кому мешал?...

okalachev, 2019-04-26:
Если переключать Клевер в режим Wi-Fi Клиента – в старой конфигурации все ломалось.

copterspace, 2019-04-26:
А зачем Клевер - в режим WiFi клиента? 
Один общий ROS на несколько дронин чтоль сделать?
На Хакатоне у меня Жужа была Wifi-сервером, Клевер - WiFi клиентом. Но внутри отдельно у Жужи и отдельно у Клевера свой ROS+Клевер внутри прекрасно жил... На старой конфигурации, с ROS_IP, не помню, чтоб чё-то ломалось...

sfalexrog, 2019-04-26:
Да, в текущей версии `ROS_IP` заменён на `ROS_HOSTNAME`, это позволяет относительно безболезненно переводить Raspberry в режим клиента для Wi-Fi и подключаться через Ethernet. В `ROS_HOSTNAME` записано имя `raspberrypi.local`, которое должно разрешаться через mDNS (например, с помощью Avahi). При этом разрешение имён через mDNS будет работать только в текущей подсети коптера, то есть в вашем случае надо настроить сетевой адаптер виртуальной машины на работу в режиме bridge, чтобы при подключении к сети коптера ВМ получала ip-адрес вида 192.168.11.*.

Возможно, придётся дополнительно настроить разрешение имён через avahi, в некоторых версиях дистрибутивов оно по умолчанию не работает.

okalachev, 2019-06-23:
@copterspace, как ситуация в 17-м образе? Все ок?

copterspace, 2019-06-24:
Я, честно говоря, не пробовал, надо? 
Из предыдущей переписки, изучив вопрос, для себя я пришёл к выводу, что, поскольку все mDNSы всё равно работают поверх DHCP - то простой и правильный путь - это ROS_IP + DHCP, a всё остальное - лишние "навороты в зоопарке". 
К тому в своём сетапе и вернулся, и на этом остановился пока что.

okalachev, 2019-06-25:
Ладно, я пока закрою issue, потому что пока непонятно, что по нему делать.

copterspace, 2019-11-07:
Тестирую 18-й клевер. Подключил виртуалку в режиме bridge, получил ip-адрес 192.168.11.117 в сети коптера. 
Прописал на виртуалке в /etc/avahi/avahi-daemon.conf:
```
host-name=copterspaceVM
domain-name=local
```
Присвоил на виртуалке ROS_MASTER_URI=http://192.168.11.1:11311 , ROS_HOSTNAME=copterspaceVM.local

Но это не помогло. Результат ровно такой же: rostopic list - показывает полный список, а rostopic echo /mavros/state - молчит на виртуалке и отражается только в ssh сеансе.

Прописал на Rapsberry в /etc/avahi/avahi-daemon.conf:
```
host-name=clever-4727
domain-name=local
```
Также не дало никакого эффекта.

Можно ли более подробно пояснить, что имелось в виду под "дополнительно настроить разрешение имён через avahi"? На виртуалке стоит коробочная Ubuntu 16.4 LTS x64.

okalachev, 2019-11-07:
Чтобы все работало, виртуалка должна пинговаться с Клевера по тому имени, которое назначается как ROS_HOSTNAME:

```bash
ping copterspaceVM.local
```

Хотя гораздо проще в данном случае просто запускать так:

```bash
ROS_MASTER_URI=http://192.168.11.1:11311 ROS_HOSTNAME= 192.168.11.117 cmd
```

copterspace, 2019-11-08:
```ROS_HOSTNAME=192.168.11.117```  эффекта также не дало.
```ROS_MASTER_URI=http://192.168.11.1:11311 ROS_HOSTNAME=192.168.11.117 rostopic list``` - кажет список топиков ОК, а  ```ROS_MASTER_URI=http://192.168.11.1:11311 ROS_HOSTNAME=192.168.11.117 rostopic echo /mavros/state``` - по-прежнему молчит.

Буду разбираться с разрешением имён в avahi, в нагугленных инструкциях пока только "запустите демона avahi - и будет вам счастье", демоны везде запущены, но по именам так и не видят друг друга...

okalachev, 2019-11-10:
>Т.е., проявился чуть другой принцип - то что задано в ROS_HOSTNAME на Малине ("clever-4727.local", в нашем случае) - должно пинговаться с Убунты.

Да, это так и должно работать. Поэтому в образе для Клевера в качестве `ROS_HOSTNAME` указано `clever-XXXX.local`. Таким образом это может работать и по Wi-Fi, и по Ethernet. mDNS-имя `clever-XXXX.local` у нас резолвилось на всех Ubuntu без всяких настроек и проблем с этим не было. Более того, это имя также резволится из коробки с Мака.

closed 2019-11-12 by okalachev