CopterExpress/clover #178: Не собирается софт Clever. Tag v0.18-rc1

vilesovds, 2019-09-14:
Имеется ноутбук с 64-разрядной ubuntu 16.04 LTS
Собирал по инструкции [Сборка образа Клевера в симуляторе](https://clever.coex.tech/ru/sitl.html#%D1%81%D0%B1%D0%BE%D1%80%D0%BA%D0%B0-%D0%BE%D0%B1%D1%80%D0%B0%D0%B7%D0%B0-%D0%BA%D0%BB%D0%B5%D0%B2%D0%B5%D1%80%D0%B0-%D0%B2-%D1%81%D0%B8%D0%BC%D1%83%D0%BB%D1%8F%D1%82%D0%BE%D1%80%D0%B5)
При попытке сборки возникает **ошибка** 

> led.cpp:18:30: fatal error: led_msgs/SetLEDs.h: Нет такого файла или каталога

Полный лог сборки: https://pastebin.com/kgbPpaZf
Примечательно, что при переключении на tag v0.17 всё собирается
Прошу помощи.

okalachev, 2019-09-14:
Спасибо за багрепорт. Проблема в том, что в версии 0.18 мы стали использовать отсутствующий в репозиториях ROS пакет led_msgs. Мы также пока не собрали его под amd64 и не выложили в наши репозитории.

Этот пакет находится здесь: https://github.com/CopterExpress/ros_led/tree/master/led_msgs. Необходимо добавить его в свой `catkin_ws/src` и собрать.

vilesovds, 2019-09-14:
Ну и резюмирую
Докачал репозиторий:
```
cd ~/catkin_ws/src
git clone https://github.com/CopterExpress/ros_led.git
```
после чего собрались и  **led_msgs** и **ws281x** ну и вся сборка в целом удалась.
Спасибо, @okalachev

closed 2019-09-14 by okalachev