CopterExpress/clover #203: Будет ли работать образ/пи с arducopter

whoim2, 2020-01-23:
Будет ли работать образ/пи с arducopter на чибиос для F4, например на омнибас?

https://firmware.ardupilot.org/Copter/latest/omnibusf4pro/

Если не все будет работать, то что именно нет? Пока нет px4 в доступе

okalachev, 2020-01-23:
ChibiOS здесь не играет особой роли. Если говорить о поддержке ArduCopter, то в mavros (модуле для проброса MAVLink в ROS) есть его поддержка, то есть связь с полетным контроллером работать будет. Все остальные модули, специфичные для Клевера, не тестируются с ArduCopter. Если что-то и будет работать, то наверняка только частично.

`simple_offboard` не заведется: он заточен под режим `OFFBOARD` в PX4.

whoim2, 2020-01-23:
Спасибо, отложу тогда до появления пикса в арсенале

closed 2020-01-23 by whoim2