CopterExpress/clover #236: Симулятор с несколькими дронами

gnarwhal97, 2020-05-26:
Здравствуйте! Запускаю через px4 sitl gazebo несколько дронов. В списке запущенных ROS - нод у меня: 
/gazebo
/gazebo_gui
/map_flipped_frame
/mavros
/nodelet_manager
/rangefinder_frame
/rosout
/shell
/simple_offboard
/uav0/mavros
/uav1/mavros
/uav2/mavros
/visualization

Если использую navigate, то она применяется только к первому дрону ( /uav0/mavros). Можно ли как-то применить её к остальным двум?

okalachev, 2020-05-27:
Привет!

Нода `simple_offboard` применяет команду к тому `mavros`'у, с которым она запущена в одном неймспейсе. На каждый запущеный дрон нужен отдельный `simple_offboard`. Таким образом, имя сервиса будет выглядеть так: `/uav0/navigate`.

Необходимо также учитывать, что и `simple_offboard`, и `mavros` публикуют ROS-фреймы. При использовании одного ROS-мастера в их настройках необходимо также переименовать их соответствующим образом для каждого дрона, например, `body0`, `base_link0`, `map0`, `body1`, и т. д.

gnarwhal97, 2020-05-30:
Разобрался, Большое спасибо!

closed 2020-05-30 by okalachev