CopterExpress/clover #240: Etcher flashing error with clover_v0.20.img

Jai-GAY, 2020-06-03:
[Etcher 1.5.95](https://www.balena.io/etcher/) + [clover_v0.20.img](https://github.com/CopterExpress/clover/releases/tag/v0.20) = Checksum error
[Etcher 1.5.95](https://www.balena.io/etcher/) + [clever_v0.18.img](https://github.com/CopterExpress/clover/releases/tag/v0.18) = No warning or error

[image]

okalachev, 2020-06-03:
I believe it might be a hardware problem, the image version has nothing with the checksum check.

Please, try to flash your SD card once again.

Jai-GAY, 2020-06-03:
I had repeatedly flash, re-flash, SD card full format then flash, still, I get the error.

okalachev, 2020-06-03:
Thanks for the report! We'll definitely investigate this issue more, and maybe even switch to advising Raspberry Pi Imager instead of Etcher.

We tested with Etcher 1.4.4 and had no problems.

closed 2020-06-10 by Jai-GAY