CopterExpress/clover #241: Не отображаются координаты в телеметрии.

gnarwhal97, 2020-06-03:
Здравствуйте! При запросе телеметрии коптера командой get_telemetry в полях x,y,z выводятся значения nan. Пример вывода:
frame_id: "map"
connected: True
armed: True
mode: "OFFBOARD"
x: nan
y: nan
z: nan
lat: 47.3977665
lon: 8.5456138
alt: 47.8988990784
vx: nan
vy: nan
vz: nan
pitch: nan
roll: nan
yaw: nan
pitch_rate: -0.391697049141
roll_rate: 0.332374721766
yaw_rate: 0.0333648249507
voltage: 11.4899997711
cell_voltage: 3.82800006866

В чем может быть причина?

sfalexrog, 2020-06-03:
Я очень прошу предоставлять как можно больше подробностей при создании issue - так нам будет гораздо более понятно, что у вас за конфигурация и как вы к такой вот ситуации пришли. Поверьте, "лишней" информации не бывает, не стесняйтесь писать большие сообщения, особенно если это первое сообщение в обсуждении.

Подозреваю, что у вас продолжение того, что вы начали в #236. В таком случае, скорее всего, вы решили проблемы с фреймами и несколькими экземплярами `simple_offboard`. Я бы рекомендовал прислать дерево трансформов (tf tree), которое у вас получилось. Значения `nan` в полях `x`, `y`, `z`, `vx`, `vy`, `vz`, `pitch`, `roll`, `yaw`, скорее всего, обусловлены тем, что фрейма `map` не существует или нет трансформа в этот фрейм из фрейма дрона.

Например, если первый дрон использует фрейм `map0` вместо `map`, попробуйте запросить его телеметрию в этом фрейме.

gnarwhal97, 2020-06-03:
Да, это от части продолжение темы #236, однако я пробовал смотреть телеметрию для 1 дрона, там тоже значения nan в тех же пунктах.
Прикрепляю tr_tree
[image]

На всякий случай прикрепляю строки из clover.launch, которые я добавил, чтобы можно было контролировать все запущенные дроны.
 
    <node name="simple_offboard" pkg="clover" type="simple_offboard" output="screen" clear_params="true">
        <param name="reference_frames/body" value="map"/>
        <param name="reference_frames/base_link" value="map"/>
        <param name="reference_frames/navigate_target" value="map"/>
    </node>

    <node name="simple_offboard" pkg="clover" type="simple_offboard" output="screen" clear_params="true" ns="uav1">
        <param name="reference_frames/body1" value="map1"/>
        <param name="reference_frames/base_link1" value="map1"/>
        <param name="reference_frames/navigate_target1" value="map1"/>
    </node>

    <node name="simple_offboard" pkg="clover" type="simple_offboard" output="screen" clear_params="true" ns="uav2">
        <param name="reference_frames/body2" value="map2"/>
        <param name="reference_frames/base_link2" value="map2"/>
        <param name="reference_frames/navigate_target2" value="map2"/>
    </node>

sfalexrog, 2020-06-04:
Судя по tf tree, у вас запущены несколько экземпляров `mavros` с одинаковыми настройками фреймов. Надо, чтобы, например, `/uav1/mavros` публиковал трансформ `map1->base_link1`, `/uav2/mavros` - `map2->base_link2` и так далее. В свою очередь, у `/uav1/simple_offboard` надо поставить параметр `body_frame` в `body1`, `target_frame` - в `navigate_target1`, `setpoint`  - в `setpoint1`; у `/uav2/simple_offboard` - `body2`, дальше - по аналогии. `map_ned` тоже надо как-то переименовать в `map1_ned`, `map2_ned` (это в настройках инстансов мавроса). Публикацию `map_flipped`, наверное, вообще можно убрать.

Ещё у вас отсутствует трансформ `map->base_link` (его должен `mavros` публиковать, причём каждый - для своего дрона). Такое бывает, если `mavros` не смог подключиться к запущенному `px4`, хотя это и немного странно.

В любом случае, я вам настоятельно рекомендую сначала просто с одним дроном разобраться, у него все фреймы переименовать, ноды запихнуть в namespace, убедиться, что всё работает, и после этого уже переходить к нескольким.

gnarwhal97, 2020-06-05:
Может ли быть проблема в том, что когда я запускаю roslaunch clover sitl.launch и пытаюсь отправить какую-нибудь команду дрону (тот же navigate), то выдает ошибку:
success: False
message: "State timeout, check mavros settings"
Но если я после запуска clover sitl laucnh, запущу в отдельном окне mavros, то он выдаст ошибку, что mavors уже запущен, выключится, но тогда все начнет работать.

closed 2020-07-12 by gnarwhal97