CopterExpress/clover #246: Камера pi-v2 как opflow поведение при крене

whoim2, 2020-06-27:
Прошу прощения за возможно глупый вопрос, но в доках не нашел упоминания о таком.
Насколько я понимаю изображение дебага opflow, полоска от центра показывает направление, противоположное движению (как движется картинка).
Но при крене / тангаже картинка тоже движется, возможно ли что это распознаётся как горизонтальное перемещение, и поэтому коптер в position просто валится вбок, так как кренится и эффект усиливается ещё больше?
Поможет ли калибровка камеры и более сложный фон?
Фокус настроил как мог, но на мой взгляд, резкость не очень. Но лучше не настраивается.
Камера без допобъектива, стандартная камера pi v2. С ней вообще возможно получить результат opflow?

okalachev, 2020-06-27:
> Но при крене / тангаже картинка тоже движется, возможно ли что это распознаётся как горизонтальное перемещение

При вычислении горизонтальной скорости зафиксированные угловые скорости дрона компенсируются.

>Камера без допобъектива, стандартная камера pi v2.

Калибровка по умолчанию для камеры с широкоугольным объективом (fish eye). При использовании другой камеры необходима [калибровка](https://clover.coex.tech/ru/camera_calibration.html).

whoim2, 2020-06-27:
Спасибо, приступаю к калибовке, отпишусь.

whoim2, 2020-06-28:
Долгим был путь к калибровке камеры по разным причинам, но в итоге удалось.
Однако, для проверки хотел использовать rviz и:

> ROS_MASTER_URI=http://192.168.11.1:11311 rviz --v
[ INFO] [1593356768.084401399]: rviz version 1.13.12
[ INFO] [1593356768.085184606]: compiled against Qt version 5.9.5
[ INFO] [1593356768.085898599]: compiled against OGRE version 1.9.0 (Ghadamon)
[ INFO] [1593356768.127080515]: Forcing OpenGl version 0.
Ошибка сегментирования (стек памяти сброшен на диск)

Не подскажете, может, сталкивались? Нетбук на Атоме 250м, убунта 18.04, росмелодик по статье встал без проблем full

sfalexrog, 2020-06-28:
Подозреваю, что могут быть проблемы с OpenGL. Не очень понятно, что такое "атом 250й", желательно указать полную модель CPU (так, как это написано в `/proc/cpuinfo`, например). Проверьте, что у вас в принципе поддержка OpenGL есть (запустите `glxinfo` из пакета `mesa-utils` и смотрите на строки `direct rendering`, `OpenGL vendor string`, `OpenGL version string`), проверьте, что OpenGL в принципе работает (запустите `glxgears` из пакета `mesa-utils`).

closed 2021-08-05 by okalachev