CopterExpress/clover #249: Не работает get_telemetry в симуляции

gnarwhal97, 2020-07-12:
Здравствуйте! Так и не смог разобраться с проблемой. При вызове get_telemetry() выдает следующее:
rame_id: "map"
connected: True
armed: True
mode: "OFFBOARD"
x: nan
y: nan
z: nan
lat: 47.3977665
lon: 8.5456138
alt: 47.8988990784
vx: nan
vy: nan
vz: nan
pitch: nan
roll: nan
yaw: nan
pitch_rate: -0.391697049141
roll_rate: 0.332374721766
yaw_rate: 0.0333648249507
voltage: 11.4899997711
cell_voltage: 3.82800006866

Запускаю симулятор все таким вот launch файлом:

<?xml version="1.0"?>
<launch>
    
    
    
    <arg name="est" default="ekf2"/>
    <arg name="vehicle" default="iris"/>
    <arg name="world" default="$(find mavlink_sitl_gazebo)/worlds/empty.world"/>
    
    <arg name="gui" default="true"/>
    <arg name="debug" default="false"/>
    <arg name="verbose" default="false"/>
    <arg name="paused" default="false"/>
    
    <include file="$(find gazebo_ros)/launch/empty_world.launch">
        <arg name="gui" value="$(arg gui)"/>
        <arg name="world_name" value="$(arg world)"/>
        <arg name="debug" value="$(arg debug)"/>
        <arg name="verbose" value="$(arg verbose)"/>
        <arg name="paused" value="$(arg paused)"/>
    </include>
    
    <group ns="uav0">
        
        <arg name="ID" value="0"/>
        <arg name="fcu_url" default="udp://:14540@localhost:14580"/>
        
        <include file="$(find px4)/launch/single_vehicle_spawn.launch">
            <arg name="x" value="0"/>
            <arg name="y" value="0"/>
            <arg name="z" value="0"/>
            <arg name="R" value="0"/>
            <arg name="P" value="0"/>
            <arg name="Y" value="0"/>
            <arg name="vehicle" value="$(arg vehicle)"/>
            <arg name="mavlink_udp_port" value="14560"/>
            <arg name="mavlink_tcp_port" value="4560"/>
            <arg name="ID" value="$(arg ID)"/>
        </include>
        
        <include file="$(find mavros)/launch/px4.launch">
            <arg name="fcu_url" value="$(arg fcu_url)"/>
            <arg name="gcs_url" value=""/>
            <arg name="tgt_system" value="$(eval 1 + arg('ID'))"/>
            <arg name="tgt_component" value="1"/>
        </include>
	
	<include file="$(find clover)/launch/sitl.launch">
	</include>
    </group>
    
    <group ns="uav1">
        
        <arg name="ID" value="1"/>
        <arg name="fcu_url" default="udp://:14541@localhost:14581"/>
        
        <include file="$(find px4)/launch/single_vehicle_spawn.launch">
            <arg name="x" value="1"/>
            <arg name="y" value="0"/>
            <arg name="z" value="0"/>
            <arg name="R" value="0"/>
            <arg name="P" value="0"/>
            <arg name="Y" value="0"/>
            <arg name="vehicle" value="$(arg vehicle)"/>
            <arg name="mavlink_udp_port" value="14561"/>
            <arg name="mavlink_tcp_port" value="4561"/>
            <arg name="ID" value="$(arg ID)"/>
        </include>
        
        <include file="$(find mavros)/launch/px4.launch">
            <arg name="fcu_url" value="$(arg fcu_url)"/>
            <arg name="gcs_url" value=""/>
            <arg name="tgt_system" value="$(eval 1 + arg('ID'))"/>
            <arg name="tgt_component" value="1"/>
        </include>
	
	<include file="$(find clover)/launch/sitl.launch">
	</include>	
    </group>
    
    <group ns="uav2">
        
        <arg name="ID" value="2"/>
        <arg name="fcu_url" default="udp://:14542@localhost:14582"/>
        
        <include file="$(find px4)/launch/single_vehicle_spawn.launch">
            <arg name="x" value="0"/>
            <arg name="y" value="1"/>
            <arg name="z" value="0"/>
            <arg name="R" value="0"/>
            <arg name="P" value="0"/>
            <arg name="Y" value="0"/>
            <arg name="vehicle" value="$(arg vehicle)"/>
            <arg name="mavlink_udp_port" value="14562"/>
            <arg name="mavlink_tcp_port" value="4562"/>
            <arg name="ID" value="$(arg ID)"/>
        </include>
        
        <include file="$(find mavros)/launch/px4.launch">
            <arg name="fcu_url" value="$(arg fcu_url)"/>
            <arg name="gcs_url" value=""/>
            <arg name="tgt_system" value="$(eval 1 + arg('ID'))"/>
            <arg name="tgt_component" value="1"/>
        </include>
	
	<include file="$(find clover)/launch/sitl.launch">
	</include>
    </group>
</launch>

gnarwhal97, 2020-07-12:
<launch>
    
    
    
    <arg name="est" default="ekf2"/>
    <arg name="vehicle" default="iris"/>
    <arg name="world" default="$(find mavlink_sitl_gazebo)/worlds/empty.world"/>
    
    <arg name="gui" default="true"/>
    <arg name="debug" default="false"/>
    <arg name="verbose" default="false"/>
    <arg name="paused" default="false"/>
    
    <include file="$(find gazebo_ros)/launch/empty_world.launch">
        <arg name="gui" value="$(arg gui)"/>
        <arg name="world_name" value="$(arg world)"/>
        <arg name="debug" value="$(arg debug)"/>
        <arg name="verbose" value="$(arg verbose)"/>
        <arg name="paused" value="$(arg paused)"/>
    </include>
    
    <group ns="uav0">
        
        <arg name="ID" value="0"/>
        <arg name="fcu_url" default="udp://:14540@localhost:14580"/>
        
        <include file="$(find px4)/launch/single_vehicle_spawn.launch">
            <arg name="x" value="0"/>
            <arg name="y" value="0"/>
            <arg name="z" value="0"/>
            <arg name="R" value="0"/>
            <arg name="P" value="0"/>
            <arg name="Y" value="0"/>
            <arg name="vehicle" value="$(arg vehicle)"/>
            <arg name="mavlink_udp_port" value="14560"/>
            <arg name="mavlink_tcp_port" value="4560"/>
            <arg name="ID" value="$(arg ID)"/>
        </include>
        
        <include file="$(find mavros)/launch/px4.launch">
            <arg name="fcu_url" value="$(arg fcu_url)"/>
            <arg name="gcs_url" value=""/>
            <arg name="tgt_system" value="$(eval 1 + arg('ID'))"/>
            <arg name="tgt_component" value="1"/>
        </include>
	
	<include file="$(find clover)/launch/sitl.launch">
	</include>
    </group>
    
    <group ns="uav1">
        
        <arg name="ID" value="1"/>
        <arg name="fcu_url" default="udp://:14541@localhost:14581"/>
        
        <include file="$(find px4)/launch/single_vehicle_spawn.launch">
            <arg name="x" value="1"/>
            <arg name="y" value="0"/>
            <arg name="z" value="0"/>
            <arg name="R" value="0"/>
            <arg name="P" value="0"/>
            <arg name="Y" value="0"/>
            <arg name="vehicle" value="$(arg vehicle)"/>
            <arg name="mavlink_udp_port" value="14561"/>
            <arg name="mavlink_tcp_port" value="4561"/>
            <arg name="ID" value="$(arg ID)"/>
        </include>
        
        <include file="$(find mavros)/launch/px4.launch">
            <arg name="fcu_url" value="$(arg fcu_url)"/>
            <arg name="gcs_url" value=""/>
            <arg name="tgt_system" value="$(eval 1 + arg('ID'))"/>
            <arg name="tgt_component" value="1"/>
        </include>
	
	<include file="$(find clover)/launch/sitl.launch">
	</include>	
    </group>
    
    <group ns="uav2">
        
        <arg name="ID" value="2"/>
        <arg name="fcu_url" default="udp://:14542@localhost:14582"/>
        
        <include file="$(find px4)/launch/single_vehicle_spawn.launch">
            <arg name="x" value="0"/>
            <arg name="y" value="1"/>
            <arg name="z" value="0"/>
            <arg name="R" value="0"/>
            <arg name="P" value="0"/>
            <arg name="Y" value="0"/>
            <arg name="vehicle" value="$(arg vehicle)"/>
            <arg name="mavlink_udp_port" value="14562"/>
            <arg name="mavlink_tcp_port" value="4562"/>
            <arg name="ID" value="$(arg ID)"/>
        </include>
        
        <include file="$(find mavros)/launch/px4.launch">
            <arg name="fcu_url" value="$(arg fcu_url)"/>
            <arg name="gcs_url" value=""/>
            <arg name="tgt_system" value="$(eval 1 + arg('ID'))"/>
            <arg name="tgt_component" value="1"/>
        </include>
	
	<include file="$(find clover)/launch/sitl.launch">
	</include>
    </group>
</launch>

gnarwhal97, 2020-07-12:
<group ns="uav0">
        
        <arg name="ID" value="0"/>
        <arg name="fcu_url" default="udp://:14540@localhost:14580"/>
        
        <include file="$(find px4)/launch/single_vehicle_spawn.launch">
            <arg name="x" value="0"/>
            <arg name="y" value="0"/>
            <arg name="z" value="0"/>
            <arg name="R" value="0"/>
            <arg name="P" value="0"/>
            <arg name="Y" value="0"/>
            <arg name="vehicle" value="$(arg vehicle)"/>
            <arg name="mavlink_udp_port" value="14560"/>
            <arg name="mavlink_tcp_port" value="4560"/>
            <arg name="ID" value="$(arg ID)"/>
        </include>
        
        <include file="$(find mavros)/launch/px4.launch">
            <arg name="fcu_url" value="$(arg fcu_url)"/>
            <arg name="gcs_url" value=""/>
            <arg name="tgt_system" value="$(eval 1 + arg('ID'))"/>
            <arg name="tgt_component" value="1"/>
        </include>
	
	<include file="$(find clover)/launch/sitl.launch">
	</include>

gnarwhal97, 2020-07-12:
Запущенные ноды 
/gazebo
/gazebo_gui
/rosout
/uav0/map_flipped_frame
/uav0/mavros
/uav0/nodelet_manager
/uav0/shell
/uav0/simple_offboard
/uav1/map_flipped_frame
/uav1/mavros
/uav1/nodelet_manager
/uav1/shell
/uav1/simple_offboard
/uav2/map_flipped_frame
/uav2/mavros
/uav2/nodelet_manager
/uav2/shell
/uav2/simple_offboard

okalachev, 2020-07-13:
А что с frame'ами mavros'а, simple_offboard'а и прочего? Они переименованы для каждого из дронов?

gnarwhal97, 2020-07-13:
Нет, ничего не переименовывал. Просто добавил запуск кловера для каждого дрона в стандарный launch px4 для нескольких дронов. 
До этого пробовал переименовывать, но ничего не выходило. Может есть пример работающего launch файла хотя бы для одного дрона с правильными настройками, чтобы все работало?

okalachev, 2020-07-13:
Так один дрон работает, так что не очень понял вопрос. Более глубоко сейчас симуляторные дела прорабатываются в ветке https://github.com/CopterExpress/clover/tree/clover_description.

gnarwhal97, 2020-07-13:
Проблема в том, у меня даже с запуском 1 дрона, не работала get_telemetry()

gnarwhal97, 2020-07-14:
> Так один дрон работает, так что не очень понял вопрос. Более глубоко сейчас симуляторные дела прорабатываются в ветке https://github.com/CopterExpress/clover/tree/clover_description.

Собрал пакет кловера из этой ветки. Запускаю roslaunch clover_simulation simulator.launch, но команды кловера не работают, например на команду navigate пишет:
success: False
message: "State timeout, check mavros settings"

okalachev, 2020-07-18:
Это значит, что нет соединения между mavros и PX4. Обратите внимание, что в более свежих версиях PX4 поменялся порт для UDP-соединения. Чтобы учесть это изменение необходимо поменять аргумент `fcu_conn` на `sitl`. Возможно, проблема в этом.

Stanislav1987, 2020-11-03:
У меня тоже возникала такая ошибка. Я сменил кабель-улитку на другой и все заработало

closed 2021-10-15 by okalachev