CopterExpress/clover #261: Why we need a Velocity estimation during automate flight?

kids0cn, 2020-07-28:
I try to run an example program "flight.py", but it failed, it armed,but cannot takeoff。 Then I run the selfcheck.py, it tell me I need a 
velocity estimatiom, it's very hard to find this one in our contry。 Is it necessary for drone to fly?  you know, we have accelerator 。

okalachev, 2020-07-28:
You need modified PX4 version for stable takeoff: https://clover.coex.tech/en/firmware.html. Do you use it?

Also, can you please provide the output of your `selfcheck.py` run?

kids0cn, 2020-07-31:
Output  like this。
Firmware is the 1.10.0 stable

We cannot get the velocity estimation all the time and the local position sometimes。

`[WARN] [1596174460.936834]: FCU: not running Clover PX4 firmware, https://clover.coex.tech/firmware
[INFO] [1596174460.968883]: FCU: selected estimator: EKF2
[INFO] [1596174460.997979]: FCU: board rotation: no rotation
[INFO] [1596174462.683207]: IMU: OK
[WARN] [1596174463.725685]: Local position: no local position
[WARN] [1596174464.772381]: Velocity estimation: no velocity estimation
[INFO] [1596174464.963563]: Global position (GPS): OK
[INFO] [1596174465.061822]: Preflight status: OK
[WARN] [1596174466.097628]: Main camera: main_camera: no images (is the camera connected properly?)
[INFO] [1596174466.189875]: ArUco markers: aruco_detect is not running
[INFO] [1596174466.238043]: Simple offboard node: OK
[WARN] [1596174466.789452]: Optical flow: no optical flow data (from Raspberry)
[WARN] [1596174468.853867]: Vision position estimate: no VPE or MoCap messages
[INFO] [1596174469.237873]: Rangefinder: EKF2_HGT_MODE = Range sensor, operating over flat surface
[INFO] [1596174469.261750]: Rangefinder: EKF2_RNG_AID = 1, range sensor aiding enabled
[WARN] [1596174469.315333]: RPi health: system throttled to prevent damage
[WARN] [1596174469.327584]: RPi health: your system is susceptible to throttling
[WARN] [1596174469.338483]: RPi health: not enough power for onboard computer, flight inadvisable
[WARN] [1596174469.348658]: RPi health: power supply cannot provide enough power
[INFO] [1596174469.639233]: CPU usage: OK`

on the top of the informations,I get these:

`[INFO] [1596174459.502763]: Performing selfcheck...
[INFO] [1596174459.511667]: Image: version: v0.20
[INFO] [1596174459.521373]: Board: Raspberry Pi 3 Model B Plus Rev 1.3
[WARN] [1596174459.641568]: clover.service: ERROR: ros.cv_camera./main_camera: failed to open device... do nothing: device_path /dev/video0 cannot be opened
[WARN] [1596174459.652020]: clover.service: ERROR: ros.mavros.fcu: FCU: REJECT ALTITUDE CONTROL (7)
[WARN] [1596174459.661849]: clover.service: [ WARN] [1596172713.130567308]: ros.mavros.time: TM : RTT too high for timesync: 1612.09 ms.
[WARN] [1596174459.671535]: clover.service: [ WARN] [1596172713.135793819]: ros.mavros.time: TM : RTT too high for timesync: 1017.34 ms.
[WARN] [1596174459.682995]: clover.service: [ WARN] [1596172713.137202048]: ros.mavros.global_position: GP: No GPS fix
[WARN] [1596174459.692376]: clover.service: [ WARN] [1596172714.146018193]: ros.mavros.cmd: CMD: Unexpected command 520, result 0
[WARN] [1596174459.701600]: clover.service: ERROR: ros.mavros.fcu: FCU: REJECT POSITION CONTROL (39)
[WARN] [1596174459.710852]: clover.service: [ WARN] [1596172727.383688605]: ros.mavros.param: PR: request param #191 timeout, retries left 2, and 27 params still missing
[WARN] [1596174459.720429]: clover.service: [ WARN] [1596172743.947901307]: ros.mavros.global_position: GP: No GPS fix
[WARN] [1596174459.730144]: clover.service: [ WARN] [1596172752.939012762]: ros.mavros.cmd: CMD: Unexpected command 520, result 0
[WARN] [1596174459.739588]: clover.service: [ WARN] [1596172978.002053092]: ros.mavros.fcu: FCU: Preflight: GPS Horizontal Pos Drift too high
[WARN] [1596174459.748966]: clover.service: ERROR: ros.mavros.fcu: FCU: Arming denied! Press safety switch first (4)
[WARN] [1596174459.759217]: clover.service: ERROR: ros.mavros.fcu: FCU: Not arming: Preflight checks failed (4)
[WARN] [1596174459.769164]: clover.service: [ WARN] [1596172984.368013767]: ros.mavros.cmd: CMD: Unexpected command 176, result 0
[WARN] [1596174459.780406]: clover.service: ERROR: ros.mavros.fcu: FCU: No range data received, loitering. (18)
[WARN] [1596174459.790586]: clover.service: [ WARN] [1596172996.044470065]: ros.mavros.time: TM : RTT too high for timesync: 125.82 ms.
[WARN] [1596174459.800323]: clover.service: [ WARN] [1596173002.225821677]: ros.mavros.time: TM : RTT too high for timesync: 2107.24 ms.
[WARN] [1596174459.809572]: clover.service: [ WARN] [1596173002.230402510]: ros.mavros.time: TM : RTT too high for timesync: 1511.99 ms.
[WARN] [1596174459.818928]: clover.service: [ WARN] [1596173002.234038187]: ros.mavros.time: TM : RTT too high for timesync: 915.47 ms.
[WARN] [1596174459.829198]: clover.service: ERROR: ros.mavros.time: TM : Time jump detected. Resetting time synchroniser. (10)
[WARN] [1596174459.838777]: clover.service: [ WARN] [1596173040.779280152]: ros.mavros.fcu: FCU: Preflight: GPS Hor Speed Drift too high
[WARN] [1596174459.848265]: clover.service: [ WARN] [1596173042.316103849]: ros.mavros.cmd: CMD: Unexpected command 176, result 0
[WARN] [1596174459.858975]: clover.service: ERROR: ros.mavros.fcu: FCU: Autonomy off! Returned control to pilot (26)
[WARN] [1596174459.868593]: clover.service: [ WARN] [1596173060.079087645]: ros.mavros.cmd: CMD: Unexpected command 176, result 0
[WARN] [1596174459.877826]: clover.service: ERROR: ros.mavros.fcu: FCU: Failsafe enabled: no datalink (23)
[WARN] [1596174459.887688]: clover.service: [ WARN] [1596173072.036936442]: ros.mavros.cmd: CMD: Unexpected command 176, result 0
[WARN] [1596174459.896833]: clover.service: ERROR: ros.mavros.fcu: FCU: Collision Warning (12)
[WARN] [1596174459.906369]: clover.service: [ WARN] [1596173088.995235706]: ros.mavros.cmd: CMD: Unexpected command 176, result 0
[WARN] [1596174459.915512]: clover.service: [ WARN] [1596173140.180168499]: ros.mavros.cmd: CMD: Unexpected command 176, result 0
[... cut ...]
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/topics.py", line 750, in _invoke_callback
    cb(msg)
  File "/home/pi/catkin_ws/src/clover/clover/src/selfcheck.py", line 115, in mavlink_message_handler
    mav_msg = link.decode(mav_bytes_msg)
  File "/usr/local/lib/python2.7/dist-packages/pymavlink/dialects/v10/ardupilotmega.py", line 12451, in decode
    raise MAVError('invalid MAVLink CRC in msgID %u 0x%04x should be 0x%04x' % (msgId, crc, crc2.crc))
MAVError: invalid MAVLink CRC in msgID 126 0x04c8 should be 0xed9c`

okalachev, 2020-08-03:
The problem is that your FC doesn't obtain the local position of the drone. This happens due to incorrect configuration of the FC.

You can start with our fork of PX4 (https://clover.coex.tech/en/firmware.html) and our configuration guides.

Other problem is that the main camera connected to the RPi is not detected. You can troubleshoot it using our articles (https://clover.coex.tech/en/camera.html#troubleshooting).

Third problem is throttling of the RPi, which can be caused by not enough power to it.

kids0cn, 2020-08-03:
Thank you for your help。 I have change the firmware  to your version。
When I run the selfcheck.py  everything  seems all right， I  got  "Local position"， I got "velocity esitimator" ，I got "main camerra working "，and the rangefinder。 and I can see the picture through  192.168.11.1:8080 optical flow debug page。

but when take off the drone。and change to "position"，the drone doesn't stay at the point, and will fly to anywhere randomly。

what should I do to fix the problem?

okalachev, 2020-08-03:
Is camera orientation set correctly?

kids0cn, 2020-08-04:
I set the camera  like what it looks like  in the handbook 。 Here is a picture, the camera towards the floor。Attached at the back of the drone。
[image]

kids0cn, 2020-08-06:
Hi, for using T265 camera, I change the system from Clover image to Ubuntu Mate 18.04 . How to do to make a clover.service?

okalachev, 2020-08-06:
You should properly set the orientation of the camera (including the "yaw") like described here https://clover.coex.tech/en/camera_setup.html#frame.

lupixi, 2021-03-14:
Hello, I used clover  4 kit with clover_v0.21.img and v1.8.2-clover.13 firmware, but it always "ERROR: ros.mavros.fcu: FCU: [lpe] vision position timeout". Could you please share about possible solutions for this problem? Since the drone was crazy flying while changing to simple offboard mode.

selfcheck.py information as follows:
[image]

okalachev, 2021-03-14:
Hi, answered in your e-mail. Please address Telegram support chat (https://t.me/COEXHelpdesk) for quicker help!

closed 2022-03-13 by okalachev