CopterExpress/clover #266: Can't find ~/catkin_ws/src/clover/deploy/clover.service

kids0cn, 2020-08-06:
Hi, I want to start the clover.service by systemctl, but I can't find the folder called deploy in the path。

kids0cn, 2020-08-06:
and another question。What should I do to access the camera by  website just like using a clover firmware?

okalachev, 2020-08-06:
Hi.

`clover.service` is now located at `~/catkin_ws/src/clover/builder/assets/clover.service`

Note, that it's basically intended for using with Raspbian, so the default username is `pi`. Change the username to yours if you want to use it.

The easiest way to start clover services (include mentioned `web_video_server`) is:

```
roslaunch clover clover.launch
```

kids0cn, 2020-08-06:
I find a folder called builder,and I change the script to install all the packages and software on my raspi with Ubuntu mate 18.04,and 
finally I start the node of clover。 but  when I try to run "rosrun clover selfcheck.py", I got a problem like this：

`drone@clover-1286:~$ rosrun clover selfcheck.py
Traceback (most recent call last):
  File "/home/drone/catkin_ws/src/clover/clover/src/selfcheck.py", line 21, in <module>
    import tf2_ros
  File "/opt/ros/melodic/lib/python2.7/dist-packages/tf2_ros/__init__.py", line 39, in <module>
    from .buffer_interface import *
  File "/opt/ros/melodic/lib/python2.7/dist-packages/tf2_ros/buffer_interface.py", line 32, in <module>
    import roslib; roslib.load_manifest('tf2_ros')
  File "/opt/ros/melodic/lib/python2.7/dist-packages/roslib/launcher.py", line 64, in load_manifest
    sys.path = _generate_python_path(package_name, _rospack) + sys.path
  File "/opt/ros/melodic/lib/python2.7/dist-packages/roslib/launcher.py", line 97, in _generate_python_path
    m = rospack.get_manifest(pkg)
  File "/usr/lib/python2.7/dist-packages/rospkg/rospack.py", line 171, in get_manifest
    return self._load_manifest(name)
  File "/usr/lib/python2.7/dist-packages/rospkg/rospack.py", line 215, in _load_manifest
    retval = self._manifests[name] = parse_manifest_file(self.get_path(name), self._manifest_name, rospack=self)
  File "/usr/lib/python2.7/dist-packages/rospkg/manifest.py", line 414, in parse_manifest_file
    _static_rosdep_view = init_rospack_interface()
  File "/usr/lib/python2.7/dist-packages/rosdep2/rospack.py", line 59, in init_rospack_interface
    lookup = _get_default_RosdepLookup(Options())
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 134, in _get_default_RosdepLookup
    verbose=options.verbose)
  File "/usr/lib/python2.7/dist-packages/rosdep2/sources_list.py", line 594, in create_default
    sources = load_cached_sources_list(sources_cache_dir=sources_cache_dir, verbose=verbose)
  File "/usr/lib/python2.7/dist-packages/rosdep2/sources_list.py", line 552, in load_cached_sources_list
    with open(cache_index, 'r') as f:
IOError: [Errno 13] Permission denied: '/home/drone/.ros/rosdep/sources.cache/index'`

okalachev, 2020-08-06:
Looks like you have problems with ROS installation, because `import tf2_ros` doesn't work.

Maybe you used `root` for some rosdep manipulations so it made `rosdep` directory unaccessible to normal user. If so, restore owner of `/home/drone/.ros/rosdep/` to your current user using `chown`.

kids0cn, 2020-08-06:
Thanks a lot。 I have fixed the problems
I found some scripts in the folder  named "build" ,such as "image_init.sh","image_ros.sh" 。 I change  the to install all the softwares in my rasp with ubuntu mate 18.04。 Now it looks like working fine。 The question above is caused by  run the rosdep  update by "sudo"。 

Now ， without connect to my drone and connect with rangefinder and any other sensors。 I run the selfcheck.py。it looks like
this。  Except the error messages  caused by missing  connecting with FUC,sensors。  Is there any other problems?

`drone@clover-1286:~$ rosrun clover selfcheck.py
[INFO] [1596720185.811929]: Performing selfcheck...
[INFO] [1596720185.822232]: Image: no /etc/clover_version file, not the Clover image?
[INFO] [1596720185.828335]: Board: Raspberry Pi 3 Model B Plus Rev 1.3
[WARN] [1596720185.969915]: clover.service: Can't open /dev/mem: Permission denied
[WARN] [1596720185.980944]: clover.service: FATAL: ros.ws281x: [ws281x] native library init failed: mmap() failed
[WARN] [1596720185.991249]: clover.service: FATAL: ros.mavros: FCU: DeviceError:serial:open: No such file or directory (1274)
[WARN] [1596720185.997635]: clover.service: [led-16] process has died [pid 12530, exit code 1, cmd /home/drone/catkin_ws/devel/lib/ws281x/ws281x_node __name:=led __log:=/home/drone/.ros/log/f5fb20de-d7e1-11ea-9ed7-b827eb94cd61/led-16.log].
[WARN] [1596720186.004026]: clover.service: log file: /home/drone/.ros/log/f5fb20de-d7e1-11ea-9ed7-b827eb94cd61/led-16*.log
[WARN] [1596720186.010622]: clover.service: FATAL: ros.vl53l1x: VL53L1X: Can't start measurement: error -13
[WARN] [1596720186.016932]: clover.service: ERROR: ros.cv_camera./main_camera: failed to open device... do nothing: device_path /dev/video0 cannot be opened
[INFO] [1596720186.023118]: Network: OK
[WARN] [1596720189.058353]: FCU: no MAVROS state (check wiring)
[WARN] [1596720190.091735]: IMU: no IMU data (check flight controller calibration)
[WARN] [1596720191.121400]: Local position: no local position
[WARN] [1596720192.158438]: Velocity estimation: no velocity estimation
[INFO] [1596720193.199066]: Global position (GPS): no global position
[WARN] [1596720199.209358]: Preflight status: no data from FCU
[WARN] [1596720200.241538]: Main camera: main_camera: no images (is the camera connected properly?)
[INFO] [1596720200.295705]: ArUco markers: aruco_detect is not running
[INFO] [1596720200.321544]: Simple offboard node: OK
[WARN] [1596720200.849217]: Optical flow: no optical flow data (from Raspberry)
[WARN] [1596720202.932519]: Vision position estimate: no VPE or MoCap messages
[WARN] [1596720207.062529]: Rangefinder: no rangefinder data from Raspberry
[WARN] [1596720211.111230]: Rangefinder: no rangefinder data from PX4
[WARN] [1596720211.139386]: RPi health: your system is susceptible to throttling
[WARN] [1596720211.145864]: RPi health: power supply cannot provide enough power
[INFO] [1596720211.409880]: CPU usage: OK
[WARN] [1596720211.459439]: Boot duration: long Raspbian boot duration: 50.354s (systemd-analyze for analyzing)`

okalachev, 2020-08-06:
Well, without connection to all the sensors this tools can't say much.

okalachev, 2020-08-06:
@sfalexrog, I don't understand by `/dev/mem` is needed for `clover.service` checks BTW.

sfalexrog, 2020-08-06:
> @sfalexrog, I don't understand why `/dev/mem` is needed for `clover.service` checks BTW.

That's an error from `ws281x` native library (which is just a part of the log), not from the `clover.service` checks themselves.

closed 2021-05-13 by okalachev