CopterExpress/clover #291: clover: Cannot locate rosdep definition for [led_msgs]

eremteknoloji, 2020-12-14:
I get the following error with rosdep install -y --from-paths src --ignore-src command. How to fix this error. I will be glad if you help. I use full desktop kinetic on rpi 3 with ubuntu mate 16.04. 

DEPRECATION: Python 2.7 reached the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 is no longer maintained. pip 21.0 will drop support for Python 2.7 in January 2021. More details about Python 2 support in pip can be found at https://pip.pypa.io/en/latest/development/release-process/#python-2-support pip 21.0 will remove support for this functionality.
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
clover: Cannot locate rosdep definition for [led_msgs]
clover_simulation: Cannot locate rosdep definition for [led_msgs]

okalachev, 2020-12-14:
Hi! The `led_msgs` package is located in https://github.com/CopterExpress/ros_led repository, so please clone to your Catkin workspace from there.

okalachev, 2020-12-14:
>Thanks, I found the ros_led pack. but typing sudo chown root: root $ (catkin_find led_msgs led_msgs_node) chown: missing operand after 'root: root'

There should not be a space after the colon.

eremteknoloji, 2020-12-14:
Hi Oleg,

new error, How can i make for it. Thanks so much.
erem@erem-desktop:~/catkin_ws$ sudo chown root: root $(catkin_find led_msgs
led_msgs_node)
chown: cannot access 'root': No such file or directory

Oleg Kalachev <notifications@github.com>, 14 Ara 2020 Pzt, 17:50 tarihinde
şunu yazdı:

> Thanks, I found the ros_led pack. but typing sudo chown root: root $
> (catkin_find led_msgs led_msgs_node) chown: missing operand after 'root:
> root'
>
> There should not be a space after the colon.

okalachev, 2020-12-14:
The correct commands for running ws281x without sudo are:

```bash
sudo chown root:root $(catkin_find ws281x ws281x_node)
sudo chmod +s $(catkin_find ws281x ws281x_node)
```

Not:

```bash
sudo chown root: root $(catkin_find led_msgs led_msgs_node)
```

eremteknoloji, 2020-12-14:
Hello again Oleg,

I'll do a test on clover. I need this. I am using Ubuntu mate. I also wrote
the last commands you sent. but not tol. Please I would be grateful if you
could help. Thanks for all your help.

Oleg Kalachev <notifications@github.com>, 14 Ara 2020 Pzt, 18:04 tarihinde
şunu yazdı:

> The correct commands for running ws281x without sudo are:
>
> sudo chown root:root $(catkin_find ws281x ws281x_node)
> sudo chmod +s $(catkin_find ws281x ws281x_node)
>
> Not:
>
> sudo chown root: root $(catkin_find led_msgs led_msgs_node)

okalachev, 2020-12-14:
So what is the problem for now?

eremteknoloji, 2020-12-14:
erem@erem-desktop:~/catkin_ws$ sudo chown root: root $(catkin_find led_msgs
led_msgs_node)
chown: cannot access 'root': No such file or directory

erem@erem-desktop:~$ ls -ld .. /erem/home
drwxr-xr-x 3 root root 4096 Kas 17 21:23 ..

Oleg Kalachev <notifications@github.com>, 14 Ara 2020 Pzt, 18:33 tarihinde
şunu yazdı:

> So what is the problem for now?

okalachev, 2020-12-14:
The correct commands for running ws281x node without sudo are:

```bash
sudo chown root:root $(catkin_find ws281x ws281x_node)
sudo chmod +s $(catkin_find ws281x ws281x_node)
```

Not:

```bash
sudo chown root: root $(catkin_find led_msgs led_msgs_node)
```

eremteknoloji, 2020-12-14:
Hello Oleg,

I fixed the problem.

sudo chown user $ (catkin_find led_msgs)

Thank you for all your help.

finally how can I configure for ardupilot.

Oleg Kalachev <notifications@github.com>, 14 Ara 2020 Pzt, 18:47 tarihinde
şunu yazdı:

> The correct commands for running ws281x node without sudo are:
>
> sudo chown root:root $(catkin_find ws281x ws281x_node)
> sudo chmod +s $(catkin_find ws281x ws281x_node)
>
> Not:
>
> sudo chown root: root $(catkin_find led_msgs led_msgs_node)

okalachev, 2020-12-14:
Mainly this packages target PX4. Some feature would work with ardupilot, but only some of them.

To connect to the autopilot though USB, basically you should set the correct serial device path into here `https://github.com/CopterExpress/clover/blob/master/clover/launch/mavros.launch#L15`.

eremteknoloji, 2020-12-23:
Hello Oleg,
How can I run only the optical _flow.cpp file from Clover with runros.
There is a relationship with nodelet but I couldn't understand.

Oleg Kalachev <notifications@github.com>, 14 Ara 2020 Pzt, 19:31 tarihinde
şunu yazdı:

> Mainly this packages targets PX4 autopilot. Some feature would work with
> ardupilot, but only some of them.
>
> To connect to the autopilot though USB, basically you should set the
> correct serial device path into here
> https://github.com/CopterExpress/clover/blob/master/clover/launch/mavros.launch#L15

okalachev, 2023-06-13:
Obsolete.

closed 2023-06-13 by okalachev