CopterExpress/clover #296: Hi I need some assistance can it work with ardupilot

CoderLogy, 2021-01-06:
Pls tell me does it work with ardupilot

okalachev, 2021-01-07:
Hi. We don't support ArduPilot specifically. This packages target PX4 primarily.

But some of the features (like getting telemetry and others) might work.

CoderLogy, 2021-01-07:
> Hi. We don't support ArduPilot. These packages target PX4 primarily.

So d u know something that would work with ardupilot like a service that u
made is there any service like same pls tell  pls tell me I need it

okalachev, 2021-01-12:
Some of features will work. mavros package itself support Ardupilot, for example.

CoderLogy, 2021-01-15:
> Some of features will work. mavros package itself supports Ardupilot, for

Can u pls tel me how to program drone ardupilot with opencv rpi pls help me

closed 2021-02-22 by CoderLogy