CopterExpress/clover #328: Problem installing the simulation environment

FabianSchuetze, 2021-05-12:
I would like to install the simulation environment, as outlines in the [Native Setup](https://clover.coex.tech/en/simulation_native.html) description, but the process unfortunately fails. In particular, the error message I get at the very end while building the catkin workspace: 
```Shell
gz: symbol lookup error: /usr/lib/x86_64-linux-gnu/libgazebo_common.so.9: undefined symbol: _ZN8ignition10fuel_tools12ClientConfig12SetUserAgentERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE
sitl_gazebo/CMakeFiles/sdf.dir/build.make:62: recipe for target '/root/catkin_ws/src/sitl_gazebo/models/iris/iris.sdf' failed
make[2]: *** [/root/catkin_ws/src/sitl_gazebo/models/iris/iris.sdf] Error 127
make[2]: *** Deleting file '/root/catkin_ws/src/sitl_gazebo/models/iris/iris.sdf'
CMakeFiles/Makefile2:2137: recipe for target 'sitl_gazebo/CMakeFiles/sdf.dir/all' failed
make[1]: *** [sitl_gazebo/CMakeFiles/sdf.dir/all] Error 2
```
I could follow all the prior installation steps without any problems. I am trying to install the simulation inside a docker container with the following build instructions: 
```Shell
FROM nvidia/cudagl:10.0-devel-ubuntu18.04

ARG DEBIAN_FRONTEND=noninteractive

RUN apt-get update 
RUN apt-get install -y --no-install-recommends \
    apt-utils git curl vim unzip openssh-client wget \
    build-essential cmake \
    libopenblas-dev \
    libglib2.0-0 \
    libsm6 \
    libxext6 \
    libxrender-dev \
    python-pip\
    lsb-release

RUN sh -c 'echo "deb http://packages.ros.org/ros/ubuntu $(lsb_release -sc) main" > /etc/apt/sources.list.d/ros-latest.list'

RUN apt-key adv --keyserver 'hkp://keyserver.ubuntu.com:80' --recv-key C1CF6E31E6BADE8868B172B4F42ED6FBAB17C654

RUN apt-get update && apt-get install -y --no-install-recommends\
    ros-melodic-desktop-full\
    python-rosdep\
    python-jinja2 

RUN rosdep init
RUN rosdep update
RUN echo "source /opt/ros/melodic/setup.bash" >> ~/.bashrc
RUN source ~/.bashrc
```  
Does anybody know what might be the reason for gazebo related problem while building the catkin workspace? I am grateful for any tips or suggestions!

FabianSchuetze, 2021-05-12:
I could solve the issue by following the advice here https://answers.gazebosim.org//question/25098/gazebo-symbol-lookup-error-ignition-math-libignition-issue-after-upgrade-from-90-to-913/ and then deleting the build folders of  the catkin worspace again

closed 2021-05-12 by FabianSchuetze

okalachev, 2021-05-13:
Hi, sorry, missed the issue.

FIY, we have a [Telegram chat](tg://resolve?domain=COEXHelpdesk), where help could be more operative.