CopterExpress/clover #329: Can't launch clover simulation

FabianSchuetze, 2021-05-12:
When I run the launch file for the clover simulation `roslaunch clover_simulation simulator.launch` as per the [Native Setup](https://clover.coex.tech/en/simulation_native.html) description, I get the following error:
```Shell
RLException: while processing /root/catkin_ws/src/clover/clover/launch/clover.launch:
included file [/root/catkin_ws/src/clover/clover/launch/mavros.launch] requires the 'prefix' arg to be set
```
Does somebody know what I could do to address the problem? I am grateful for any hints or suggestions!

okalachev, 2021-05-12:
Hi, thanks for the catch!

[Fixed](https://github.com/CopterExpress/clover/commit/c0d9bd7ef0a7b0a5d7282142ca6c0d830447b0c7) the problem, so please pull the repo and try it out again.

By the way, the stable working version of the simulator is available via a [virtual machine image](https://clover.coex.tech/en/simulation_vm.html).

FabianSchuetze, 2021-05-13:
Thank you so much for your reply and for addressing the issue!

closed 2021-05-13 by FabianSchuetze