CopterExpress/clover #332: Failed to load plugin libgazebo_multirotor_base_plugin.so

suresh-guttikonda, 2021-05-24:
Hi i wad following the instructions to setup clover from [simulation_native.md](https://github.com/CopterExpress/clover/blob/master/docs/en/simulation_native.md) but when run following i am facing following error 'gazebo plugins not found'. Could you please help me to fix the issue
```
catkin_ws$ source devel/setup.bash
catkin_ws$ source /usr/share/gazebo/setup.sh
catkin_ws$ export GAZEBO_PLUGIN_PATH=$PWD:$GAZEBO_PLUGIN_PATH
catkin_ws$ roslaunch clover_simulation simulator.launch
```

```
.....
....
[Err] [Plugin.hh:178] Failed to load plugin libgazebo_multirotor_base_plugin.so: libgazebo_multirotor_base_plugin.so: cannot open shared object file: No such file or directory
[Err] [Plugin.hh:178] Failed to load plugin libgazebo_motor_model.so: libgazebo_motor_model.so: cannot open shared object file: No such file or directory
[Err] [Plugin.hh:178] Failed to load plugin libgazebo_motor_model.so: libgazebo_motor_model.so: cannot open shared object file: No such file or directory
[Err] [Plugin.hh:178] Failed to load plugin libgazebo_motor_model.so: libgazebo_motor_model.so: cannot open shared object file: No such file or directory
[Err] [Plugin.hh:178] Failed to load plugin libgazebo_motor_model.so: libgazebo_motor_model.so: cannot open shared object file: No such file or directory
[Err] [Plugin.hh:178] Failed to load plugin libgazebo_magnetometer_plugin.so: libgazebo_magnetometer_plugin.so: cannot open shared object file: No such file or directory
[Err] [Plugin.hh:178] Failed to load plugin libgazebo_barometer_plugin.so: libgazebo_barometer_plugin.so: cannot open shared object file: No such file or directory
[Err] [Plugin.hh:178] Failed to load plugin libgazebo_mavlink_interface.so: libgazebo_mavlink_interface.so: cannot open shared object file: No such file or directory
[Err] [Plugin.hh:178] Failed to load plugin libgazebo_imu_plugin.so: libgazebo_imu_plugin.so: cannot open shared object file: No such file or directory
[Wrn] [sim_leds.cpp:162] Failed to convert visual to integer: stoi, assuming 0
[Wrn] [sim_leds.cpp:129] Creating new LED controller for namespace 
[ INFO] [1621853865.491137029, 0.440000000]: LedController has started (as server)
[ INFO] [1621853865.500041617, 0.448000000]: ros.roscpp: waitForService: Service [/led/set_leds] is now available.
ERROR [mavlink] [timesync] Time jump detected. Resetting time synchroniser.
[ INFO] [1621853865.622684878, 0.556000000]: ros.clover: ready
....
...
```

suresh-guttikonda, 2021-05-24:
I am using following singularity container after i clone the required repos to `catkin_ws/src`
```
Bootstrap:docker
From:docker://ros:melodic-ros-core-bionic

%labels
AUTHOR Suresh

%files
    ./catkin_ws

%post
apt-get -y update && apt-get -y install locales wget
locale-gen de_DE.UTF-8 # change this if your are facing locale errors

    # cleanup
    rm -rf /var/lib/apt/lists/*
    apt-get clean

    # catkin tool
    sh -c 'echo "deb http://packages.ros.org/ros/ubuntu `lsb_release -sc` main" > /etc/apt/sources.list.d/ros-latest.list'
    wget http://packages.ros.org/ros.key -O - | sudo apt-key add -
    apt-get -y update
    apt-get -y install python-catkin-tools \
    			python-rosdep \
    			python-rospkg

    apt-get -y install python-pip

    # clover
    cd /catkin_ws
    rosdep init
    rosdep update
    rosdep install --from-paths src --ignore-src -y --rosdistro melodic

    # px4
    cd /catkin_ws/src
    ln -s Firmware/Tools/sitl_gazebo ./sitl_gazebo

    cd /catkin_ws/src/Firmware/Tools/setup
    ./ubuntu.sh

    # patch gazebo plugins
    cd /catkin_ws/src/Firmware/Tools/sitl_gazebo
    wget https://raw.githubusercontent.com/CopterExpress/clover_vm/master/assets/patches/sitl_gazebo.patch
    patch -p1 < sitl_gazebo.patch
    rm sitl_gazebo.patch

    # geographiclib datasets
    cd /
    wget https://raw.githubusercontent.com/mavlink/mavros/6f5bd5a1a67c19c2e605f33de296b1b1be9d02fc/mavros/scripts/install_geographiclib_datasets.sh
    chmod +x ./install_geographiclib_datasets.sh
    ./install_geographiclib_datasets.sh
    rm ./install_geographiclib_datasets.sh

    #
    apt-get -y upgrade libignition-math2
```

suresh-guttikonda, 2021-05-24:
issue2: ERROR [mavlink] Failed to start shell (-1)

rosrun terminal :
```
$ rosrun clover selfcheck.py 
[INFO] [1621860703.062938, 195.056000]: Performing selfcheck...
[INFO] [1621860703.064405, 195.056000]: Image: no /etc/clover_version file, not the Clover image?
[INFO] [1621860703.065819, 195.056000]: Board: could not open /proc/device-tree/model, not a Raspberry Pi?
[WARN] [1621860703.077773, 195.068000]: clover.service: [Errno 2] No such file or directory: '/tmp/clover.err'
[INFO] [1621860703.080211, 195.072000]: Network: OK
[INFO] [1621860709.118260, 201.104000]: FCU: no version data available from SITL
[WARN] [1621860709.120270, 201.104000]: FCU: not running Clover PX4 firmware, https://clover.coex.tech/firmware
[WARN] [1621860709.123552, 201.108000]: FCU: unable to retrieve PX4 parameter SYS_MC_EST_GROUP
[WARN] [1621860709.125474, 201.108000]: FCU: unknown selected estimator: None
[WARN] [1621860709.130758, 201.116000]: FCU: unable to retrieve PX4 parameter SENS_BOARD_ROT
[WARN] [1621860709.135466, 201.120000]: FCU: unable to retrieve PX4 parameter CBRK_USB_CHK
[WARN] [1621860709.137604, 201.120000]: FCU: set parameter CBRK_USB_CHK to 197848 for flying with USB connected
[WARN] [1621860709.152836, 201.136000]: FCU: cell voltage is not available, https://clover.coex.tech/power
[WARN] [1621860710.167894, 202.152000]: IMU: no IMU data (check flight controller calibration)
[WARN] [1621860711.181512, 203.164000]: Local position: no local position
[WARN] [1621860712.186866, 204.168000]: Velocity estimation: no velocity estimation
[INFO] [1621860713.202096, 205.184000]: Global position (GPS): no global position
[WARN] [1621860719.204677, 211.164000]: Preflight status: no data from FCU
[INFO] [1621860719.255139, 211.212000]: Main camera: camera is oriented downward, cable from camera goes backward
[INFO] [1621860719.274826, 211.232000]: ArUco markers: aruco_detect/length = 0.33 m
[INFO] [1621860719.277616, 211.236000]: ArUco markers: aruco_detector/known_tilt = map (ALL markers are on the floor)
[INFO] [1621860719.342566, 211.300000]: ArUco markers: aruco_map/known_tilt = map (marker's map is on the floor)
[INFO] [1621860719.369244, 211.328000]: ArUco markers: map has 99 markers
[WARN] [1621860720.384963, 212.340000]: ArUco markers: no map detection
[INFO] [1621860720.391343, 212.348000]: Simple offboard node: OK
[WARN] [1621860720.903304, 212.860000]: Optical flow: no optical flow data (from Raspberry)
[WARN] [1621860720.953824, 212.912000]: Vision position estimate: unable to retrieve PX4 parameter SYS_MC_EST_GROUP
[WARN] [1621860722.017035, 213.972000]: Rangefinder: unable to retrieve PX4 parameter SYS_MC_EST_GROUP
[WARN] [1621860722.028663, 213.984000]: RPi health: could not call vcgencmd binary; not a Raspberry Pi?
[WARN] [1621860722.219602, 214.176000]: CPU usage: high CPU usage (118.8%) detected: px4 (PID 819514)
[WARN] [1621860722.222123, 214.176000]: CPU usage: high CPU usage (50.0%) detected: gzserver (PID 819512)
[WARN] [1621860722.223843, 214.180000]: CPU usage: high CPU usage (37.5%) detected: gzclient (PID 819482)
[WARN] [1621860722.225522, 214.180000]: CPU usage: high CPU usage (37.5%) detected: mavros_n+ (PID 819537)
[INFO] [1621860722.241605, 214.196000]: Boot duration: OK
```

simulator.launch terminal :
```
[ INFO] [1621860577.810022341, 70.000000000]: ros.clover: publish zero
[ INFO] [1621860587.843748374, 80.000000000]: ros.clover: publish zero
ERROR [mavlink] Failed to start shell (-1)
ERROR [mavlink] Failed to start shell (-1)
[ERROR] [1621860595.485838172, 87.632000000]: ros.mavros.param: PR: Unknown parameter to get: SYS_MC_EST_GROUP
[ INFO] [1621860595.487219027, 87.636000000]: ros.clover: led: notify error
[ERROR] [1621860595.493214369, 87.640000000]: ros.mavros.param: PR: Unknown parameter to get: SENS_BOARD_ROT
[ERROR] [1621860595.497118420, 87.644000000]: ros.mavros.param: PR: Unknown parameter to get: CBRK_USB_CHK
[ WARN] [1621860597.851869908, 90.000000000]: ros.mavros.global_position: GP: No GPS fix
[ INFO] [1621860597.852051850, 90.000000000]: ros.clover: publish zero
ERROR [mavlink] Failed to start shell (-1)
ERROR [mavlink] Failed to start shell (-1)
[ERROR] [1621860607.475063267, 99.612000000]: ros.mavros.param: PR: Unknown parameter to get: SYS_MC_EST_GROUP
[ INFO] [1621860607.476181275, 99.612000000]: ros.clover: led: notify error
[ INFO] [1621860607.861460042, 100.000000000]: ros.clover: publish zero
[ERROR] [1621860608.650902086, 100.788000000]: ros.mavros.param: PR: Unknown parameter to get: SYS_MC_EST_GROUP
[ INFO] [1621860617.868487043, 110.000000000]: ros.clover: publish zero
```

suresh-guttikonda, 2021-05-24:
issue1 is fixed: there was some missing package after i fixed its working now. (i updated above def file with the fix)
issue2: i read there is no support for SITL [link](https://github.com/PX4/PX4-Autopilot/issues/12654) if you have any hack please share with me otherwise you can close this issue since i am able to run [flight.py](https://clover.coex.tech/en/programming.html)

okalachev, 2021-05-26:
Hi, sorry for the delay, you possible can get much more responsive answers in the Telegram support chat: https://t.me/COEXHelpdesk.

closed 2022-06-03 by okalachev