CopterExpress/clover #428: Can I use a clean Ubuntu image to make a control computer without downloading the img file?

yunswj, 2021-12-22:
Can I use this library and make a record on the computer myself

okalachev, 2021-12-22:
Yes, but it's not so easy to install everything required, that's why the ready image is made.

okalachev, 2021-12-22:
You can take some info in the simulator installation manual (https://clover.coex.tech/en/simulation_native.html), which is kind of a superset of the required software.

yunswj, 2021-12-22:
Your response speed is too fast, awesome! ! ! I just want to study the implementation of clover Python package in from clover import srv

okalachev, 2021-12-22:
There is no separate Clover Python library, the Python modules are generated by ROS while package building (from .srv and .msg files). Programs for Clover use `rospy` library to access Clover services and topics.

yunswj, 2021-12-22:
Let me study the .srv and .msg files of ROS. thank you for your reply

closed 2021-12-29 by okalachev