CopterExpress/clover #431: Upgrading to Px4 Firmware 1.12 with Coex Pix

mhl787156, 2022-02-01:
Hi! We've been using the clovers for a projects at our lab where we have a workflow of first testing controllers we have built in simulation (px4 sitl + gazebo) and then running on the drones. There have been a number of improvements made to the PX4 firmware since 1.8 (your release) which are now in our sitl (1.12) (and QGC) which are unfortunately not in the firmware on the drone.  (E.g. improvements to the controller, EKF, new parameters etc) 

I have attempted to run the latest PX4 firmware on the drone but it does not recognize the hardware configuration of your Pix. Therefore I wanted to ask for advice/help on how I might be able to get the latest PX4 firmware onto the clover? In particular:

- I assume there is a hardware configuration somewhere in your PX4 fork for the Coex Pix - is it (hopefully) just a case of re-compiling PX4 with this configuration available? 
- I see you made a number of changes to your fork of 1.8, would some of these have to be recreated... 

Or whether this might be a lot more challenging than I anticipated. 

Many Thanks!

okalachev, 2022-02-02:
Hi! You can run the latest PX4 firmware (1.12.3) on Clover with COEX Pix. Just use the regular `v4` version. The only hardware issue is that the internal magnetometer doesn't work, we're working on it, but haven't found the solution yet.

So you need to disable the magnetometer with `SYS_HAS_MAG` and `EKF2_MAG_TYPE`. 

We are experiencing now with the latest version/EKF2, and we having some problems with the stability of the system. But for now you can use the beta version of the image - [v0.23-rc.5](https://github.com/CopterExpress/clover/releases). And [its documentation](https://github.com/CopterExpress/clover/blob/master/docs/en/SUMMARY.md) mentions the experimental possibility of using the latest firmware and configuration for it.

The other problem is the old estimator (LPE) doesn't work with the latest firmware.

mhl787156, 2022-02-02:
Thanks for the reply! 

Ah I see, I remember last time I tried I ran into the magnetometer issue and assumed things didn't work. I will try it out and let you know if it works. 

Thanks for the link to the image beta version, we're actually using our own image right now as we're messing around with Docker and ROS2, but its interesting you are having issues with the EKF. One of the main reasons we wanted to upgrade is because we wanted to use the EKF over the LPE!

okalachev, 2022-02-04:
@mhl787156, FIY, I made a patch that fixes the magnetometer issue: https://github.com/PX4/PX4-Autopilot/pull/19125.

mhl787156, 2022-03-01:
@okalachev I have finally got round to trying the most recent version of px4 (1.13 alpha with your fix pushed) and I am struggling to get the autopilot to connect over usb serial. Mavros diagnostics shows not connected to FCU

I am not totally familiar with the PX4 parameter stack and I feel I may be missing one or two. So far I have made sure to enable TELEM2 with correct baudrate (921600) and USB link circuit breaker is set to the special value. But It will still not connect over USB - I was wondering if I might have missed something else specific to the Coex Pixs?

mhl787156, 2022-03-02:
So, I came back to try some other configurations and it appears I may have bricked my Pix as it no longer shows up on QGroundControl. I have tried re-flashing it multiple times with new PX4 and the original PX4 image with both QGC and Mission Planner, but nothing. Do you have a method of resetting the Coex Pix? I notice that there is no reset button...

okalachev, 2022-03-02:
There is a bug in the version 1.12.3, if you set LPE estimator it won't boot. 

Anyway, if the problem is related to the parameters set, then to reset them you may flash ArduPilot firmware, then check, then flash the PX4 firmware again.

mhl787156, 2022-03-08:
@okalachev I finally got it working - a bit of a summary:

- Initially started with latest v1.13 alpha branch (https://github.com/PX4/PX4-Autopilot/commit/c4bc062714a5dbf2610938d8b1243e04da46bc57) which should include your  mag PR fix. After clean EEPROM wipe, QGC successfully flashes and recognises. However noticed that mavros would not connect. After about 6-10 power cycles the Coex pix just bricks itself and QGC can no longer recognise it. I have absolutely no idea what is causing this. 
- I tried to build your PR branch directly (https://github.com/CopterExpress/Firmware/tree/mpu9250-mag) but for some reason ended up with a similar problem to the last point. Something between v1.12.3 and now is causing an issue for me. 
- Finally found way that worked, I pulled PX4-Autopilot v1.12.3 and manually added the mag change, the changes in your lpe-fix branch (https://github.com/CopterExpress/Firmware/tree/v1.12.3-lpe-fix) and clover param update branch (https://github.com/CopterExpress/Firmware/commits/clover-update). Flashed that using QGC and it successfully connects to mavros and does not brick after a few uses. 

Utterly stumped as to the delayed bricking. I might be wrong in thinking its the power cycling that is causing it fail, it might just as well be something else. But at the very least we have finally gotten it to work. 

Many Thanks for all the help :D

closed 2022-03-15 by okalachev