CopterExpress/clover #435: Simulator: LED plugin doesn't work if main_camera is disabled

okalachev, 2022-03-10:
If to disable `main_camera` in `simulator.launch`, then LED layer doesn't work.

LED plugin as the client starts:

```
LedController has started (as client)
```

But LED plugin as the server doesn't.

guisoares9, 2022-03-11:
Server or client mode for led plugin is defined by which node is processing the sim_leds plugin provided by sim_led.cpp script. When camera is **enabled**, /gazebo node process the plugin starting it as server and /gazebo_gui also process the plugin starting it as client. But when camera is **disabled**, only /gazebo_gui node process the plugin starting it as client and not advertising the /led/set_leds service and /led/state topic. We can confirm it logging the node name like:

At line ~74 of sim_led.cpp with the ROS_INFO statement:
[image]

Possible solution is setting the mode of the puglin as server for both of cases. It might not be the better solution, but it works without errors. In this way, when camera is **disabled** /gazebo_gui will work as server and advertise /led/set_leds service and /led/state topic. When camera is **enabled** /gazebo will work as server and advertise the service and topic and /gazebo_gui will also publish on /led/state topic but will not serve /led/set_leds service because it has already been served by /gazebo. We can set role variable to Role::Server equals at the line 73 in the print of sim_led.cpp.
  
[image]

okalachev, 2022-03-15:
I believe I get the reason of the problem. 

The "server part" of the plugin is running only when there is at least one camera in the world: https://answers.gazebosim.org/question/15939/visual-plugin-loaded-twice/?answer=15943#post-id-15943.

In this case, a service server was not created, and the LED layer didn't work. 

I believe I came up with a solution for this. The idea of the solution is to remove all conditions and create the service server on both client and server sides. The newer service server (if there is client and server part running) just would override the previous one, and everything should work in all the configurations.

okalachev, 2022-03-15:
This solution has a flaw though, if a client would disconnect, then it might break the service server, that runs on a client.

commit 850b49b2 (2022-03-23): Fix #435 (#438)

* Split Gazebo LED plugin to LED visual plugin and LED model plugin

* Get rid of unneeded 'Failed to convert' warning

* Minor
closed 2022-03-23 by okalachev