CopterExpress/clover #441: How use navigate function on JS

JamesFlosse, 2022-05-09:
Hi, i would like to use the navigate python function but i don't find a way to use it with rosjs?
It's possible? if yes can you tell me how?
Best regards

okalachev, 2022-05-09:
Are you asking about using ROS services in JavaScript from the browser?

There is an example on how to use the `get_telemetry` service: https://clover.coex.tech/en/javascript.html.

`navigate` service is used in the same way.

JamesFlosse, 2022-05-10:
Hi, thx for your answer.
In pythonn to hover copter above aruco you use this function:
`navigate(x=0, y=0, z=1, frame_id='aruco_107')`
I would like to do same in html with javascript.

okalachev, 2022-05-10:
Please, read the article: https://clover.coex.tech/en/javascript.html.

And use the code like:

```javascript
var navigate = new ROSLIB.Service({ ros: ros, name : '/navigate', serviceType : 'clover/Navigate' });

navigate.callService(new ROSLIB.ServiceRequest({ x: 0, y: 0, z: 1, frame_id: 'aruco_107' }), function(result) {
    console.log('Result: ' + JSON.stringify(result));
});
```

closed 2022-07-14 by okalachev