CopterExpress/clover #442: Report error with "TRANSITION_DENIED: SHUTDOWN to INIT" while calibrating sensors in Pixhawk

phtchpol, 2022-05-28:
Hi, I encountered a problem while calibrating sensor for Pixhawk. The system told me that it failed to switch state from shutdown to init even though the system doesn't allow shutdown during when Pixhawk plugged into PC via USB.
[image]

okalachev, 2022-05-30:
Can you try to perform the calibration with external power from battery? This might solve the issue.

phtchpol, 2022-06-03:
The problem still persist. Might be from outdated firmware with outdated hardware. I will try to reflash firmware again

phtchpol, 2022-06-15:
The problem solve after flashing new firmware into more up-to-dated hardware. Thank you

closed 2022-06-15 by phtchpol

Jai-GAY, 2023-01-09:
> The problem solve after flashing new firmware into more up-to-dated hardware. Thank you

May I know [which version](https://github.com/CopterExpress/Firmware/tags) did you use? I faced the same issue.