CopterExpress/clover #445: ws281x hardware revision is not supported

lylecp, 2022-06-13:
Hi 
I have installed version 0.23 on a Raspberry Pi 4 (v1.5). 
When running selfcheck.py I get the following error regarding the LED driver:

[INFO]: Image: version: v0.23
[INFO]: Board: Raspberry Pi 4 Model B Rev 1.5
[WARN]: clover.service: FATAL: ros.ws281x: [ws281x] native library init failed: Hardware revision is not supported

How can i fix this?

okalachev, 2022-06-13:
Hi, thanks for the report!

[ws281x](https://github.com/CopterExpress/ros_led/tree/master/ws281x) package should update underlaying [rpi_ws281x](https://github.com/jgarff/rpi_ws281x) library. I will do this in a couple of days.

okalachev, 2022-06-14:
@lylecp, hi, just released a new package version. Can you please test on your hardware?

You have to download here (http://packages.coex.tech/pool/main/r/ros-noetic-ws281x/ros-noetic-ws281x_0.0.13-0buster_armhf.deb) and copy it to you Raspberry Pi. Then install it like that:

```bash
sudo dpkg -i ros-noetic-ws281x_0.0.13-0buster_armhf.deb
```

Then restart Clover's services:

```bash
sudo systemctl restart clover
```

lylecp, 2022-06-18:
Thanks i will test it and let you know

closed 2022-06-29 by okalachev

lylecp, 2022-10-11:
> Can I close the issue?

Thanks yes you can close the issue.

lastavetski, 2023-03-24:
[image]