CopterExpress/clover #446: When roslaunch clover_simulation simulation.launch is run it does not spawn the gazebo urdf model

AvishkarSeth, 2022-07-05:
Error message: [WARN] [1656980174.968365, 0.000000]: wait_for_service(/gazebo/spawn_urdf_model): failed to contact, will keep trying

okalachev, 2022-07-05:
Hi, which method have you used to install the simulator?

AvishkarSeth, 2022-07-05:
> Hi, which method have you used to install the simulator?

The one in this link...
https://clover.coex.tech/en/simulation_native.html

okalachev, 2022-07-05:
Do I understand correctly the problem is that the drone doesn't appear? The warning you provided may be not the cause of it. Maybe some other errors in the output?

`/gazebo/spawn_urdf_model` service takes some time to appear, by the service caller would wait for it.

AvishkarSeth, 2022-07-05:
[image]
As per the screenshots I attached... the simulator does not launch the gazebo model and also the examples such as 'flight.py' do not work. I tried reinstalling everything thrice but same issue

okalachev, 2022-07-05:
This is only the little ending part of the output log. Can you please provide the full log? There can be errors there.

AvishkarSeth, 2022-07-05:
buntu:~/catkin_ws/src/clover/clover/examples$ roslaunch clover_simulation simulator.launch 
... logging to /home/alice/.ros/log/9a926538-fc09-11ec-abfe-6974411158b9/roslaunch-SCI-10657-Ubuntu-30850.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://127.0.1.1:33591/

SUMMARY
========

CLEAR PARAMETERS
 * /led/led_effect/
 * /main_camera_nodelet_manager/
 * /mavros/
 * /optical_flow/
 * /roswww_static/
 * /simple_offboard/
 * /vpe_publisher/

PARAMETERS
 * /drone_description: <?xml version="1....
 * /gazebo/enable_ros_network: True
 * /led/led_effect/blink_rate: 2
 * /led/led_effect/fade_period: 0.5
 * /led/led_effect/notify/acro/b: 0
 * /led/led_effect/notify/acro/g: 155
 * /led/led_effect/notify/acro/r: 245
 * /led/led_effect/notify/altctl/b: 40
 * /led/led_effect/notify/altctl/g: 255
 * /led/led_effect/notify/altctl/r: 255
 * /led/led_effect/notify/connected/effect: rainbow
 * /led/led_effect/notify/disconnected/b: 50
 * /led/led_effect/notify/disconnected/effect: blink
 * /led/led_effect/notify/disconnected/g: 50
 * /led/led_effect/notify/disconnected/r: 255
 * /led/led_effect/notify/error/b: 0
 * /led/led_effect/notify/error/effect: flash
 * /led/led_effect/notify/error/g: 0
 * /led/led_effect/notify/error/ignore: ['[lpe] vision po...
 * /led/led_effect/notify/error/r: 255
 * /led/led_effect/notify/low_battery/b: 0
 * /led/led_effect/notify/low_battery/effect: blink_fast
 * /led/led_effect/notify/low_battery/g: 0
 * /led/led_effect/notify/low_battery/r: 255
 * /led/led_effect/notify/low_battery/threshold: 3.6
 * /led/led_effect/notify/offboard/b: 250
 * /led/led_effect/notify/offboard/g: 20
 * /led/led_effect/notify/offboard/r: 220
 * /led/led_effect/notify/posctl/b: 220
 * /led/led_effect/notify/posctl/g: 100
 * /led/led_effect/notify/posctl/r: 50
 * /led/led_effect/notify/stabilized/b: 50
 * /led/led_effect/notify/stabilized/g: 180
 * /led/led_effect/notify/stabilized/r: 30
 * /led/led_effect/notify/startup/b: 255
 * /led/led_effect/notify/startup/g: 255
 * /led/led_effect/notify/startup/r: 255
 * /led/led_effect/rainbow_period: 5
 * /main_camera/main_camera_markers/scale: 3.0
 * /main_camera_nodelet_manager/num_worker_threads: 2
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 8
 * /mavros/conn/timesync_rate: 10.0
 * /mavros/distance_sensor/rangefinder/field_of_view: 0.5
 * /mavros/distance_sensor/rangefinder/frame_id: rangefinder
 * /mavros/distance_sensor/rangefinder/id: 0
 * /mavros/distance_sensor/rangefinder/orientation: PITCH_270
 * /mavros/distance_sensor/rangefinder_sub/covariance: 1
[... 325 lines cut ...]
Take off and hover 1 m above the ground
Traceback (most recent call last):
  File "/opt/ros/noetic/lib/python3/dist-packages/rospy/impl/tcpros_base.py", line 567, in connect
    self.socket.connect((dest_addr, dest_port))
ConnectionRefusedError: [Errno 111] Connection refused

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/opt/ros/noetic/lib/python3/dist-packages/rospy/impl/tcpros_service.py", line 509, in call
    transport.connect(dest_addr, dest_port, service_uri)
  File "/opt/ros/noetic/lib/python3/dist-packages/rospy/impl/tcpros_base.py", line 596, in connect
    raise TransportInitError(str(e)) #re-raise i/o error
rospy.exceptions.TransportInitError: [Errno 111] Connection refused

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "flight.py", line 19, in <module>
    navigate(x=0, y=0, z=1, frame_id='body', auto_arm=True)
  File "/opt/ros/noetic/lib/python3/dist-packages/rospy/impl/tcpros_service.py", line 442, in __call__
    return self.call(*args, **kwds)
  File "/opt/ros/noetic/lib/python3/dist-packages/rospy/impl/tcpros_service.py", line 512, in call
    raise ServiceException("unable to connect to service: %s"%e)
rospy.service.ServiceException: unable to connect to service: [Errno 111] Connection refused

okalachev, 2022-07-05:
Maybe some networking issues?

Can this machine ping itself? Do other ROS tools work, while the sim is running? (`rostopic list`, `rosservice list`, `rostopic echo <any-topic>`, etc. 

What is the output of `rosservice list`?

AvishkarSeth, 2022-07-05:
[image]
Would this help?

okalachev, 2022-07-05:
What about the list of services (`rosservice list`) and other stuff, when the sim is running? Note, you don't have to run `roscore` yourself, roslaunch will do this.

AvishkarSeth, 2022-07-05:
Hi Okalachev,
[image]
[image]

In the 'simulator.launch' file, what model is it calling? Is it 'clover.launch'? And maybe the file in the git repo is named 'clover4.launch' or something? I'm not sure...

AvishkarSeth, 2022-07-06:
Hi Okalachev,

These two links helped resolve the issue!
1. https://answers.gazebosim.org//question/21602/spawning-baxterurdf-not-possible-gazebo-doesnt-start-waiting-for-service/
2. https://forum.humanbrainproject.eu/t/waiting-for-service-gazebo-spawn-urdf-model/468
Thanks for the help!

okalachev, 2022-07-06:
Great!

closed 2022-07-06 by okalachev