CopterExpress/clover #449: ros.clover: Can't transform from aruco_XX to map

JamesFlosse, 2022-07-27:
Hi,
I have this error on Clover simulator when i try to use this js function:
`var navigate = new ROSLIB.Service({ ros: ros, name : '/navigate', serviceType : 'clover/Navigate' });
                navigate.callService(new ROSLIB.ServiceRequest({ x: 0, y: 0, z: hauteur, frame_id: 'aruco_90' }), function(result) {
                    console.log('Result: ' + JSON.stringify(result));
                });`

Can you help me to resolve it please?

PS: From this is issues https://github.com/CopterExpress/clover/issues/441#issuecomment-1122235392

JamesFlosse, 2022-07-27:
Ok i anwer myself, i need to 
`var navigate = new ROSLIB.Service({ ros: ros, name : '/navigate', serviceType : 'clover/Navigate' }); navigate.callService(new ROSLIB.ServiceRequest({ x: 0, y: 0, z: hauteur, frame_id: 'body', auto_arm=True }), function(result) { console.log('Result: ' + JSON.stringify(result)); });`
and after i can call
`var navigate = new ROSLIB.Service({ ros: ros, name : '/navigate', serviceType : 'clover/Navigate' }); navigate.callService(new ROSLIB.ServiceRequest({ x: 0, y: 0, z: hauteur, frame_id: 'aruco_90' }), function(result) { console.log('Result: ' + JSON.stringify(result)); });`

okalachev, 2022-07-27:
May be closed?

JamesFlosse, 2022-07-27:
I make this script for hover aruco marker everytime, so if aruco marker it doesn't view by the drone, aruco marker can't transform? exactly?

` function arm() {
            var navigate = new ROSLIB.Service({ ros: ros, name : '/navigate', serviceType : 'clover/Navigate' });
            navigate.callService(new ROSLIB.ServiceRequest({ x: 0, y: 0, z: 0,speed: 0.1, frame_id: 'aruco_90', auto_arm: true }), function(result) {
                console.log('Result: ' + JSON.stringify(result));
            });
        }

        function takeOff() {            
            var navigate = new ROSLIB.Service({ ros: ros, name : '/navigate', serviceType : 'clover/Navigate' });
            navigate.callService(new ROSLIB.ServiceRequest({ x: 0, y: 0, z: 1,speed: 0.5, frame_id: 'aruco_90', auto_arm: true }), function(result) {
                console.log('Result: ' + JSON.stringify(result));
            });

        }

        function land() {
            var navigate = new ROSLIB.Service({ ros: ros, name : '/land', serviceType : 'clover/Land' });
            navigate.callService(new ROSLIB.ServiceRequest({ x: 0, y: 0, z: 0,speed: 0.1, frame_id: 'aruco_90', auto_arm: true }), function(result) {
                console.log('Result: ' + JSON.stringify(result));
            });
}`

okalachev, 2022-07-27:
If the drone have seen the configured aruco map once:

- Transforms to markers that are in the map are available.
- Transforms to markers that are not in the map available only when the drone sees them.

If the drone haven't seen the map or the map is not condifured:

- Transforms to markers available only when the drone sees them.

okalachev, 2022-07-27:
In this case, like I said, `aruco_X` transform is available only when the drone sees the marker.

JamesFlosse, 2022-07-27:
Ok thank you for the precision ;)
And nice project you are make :).

closed 2022-07-27 by JamesFlosse