CopterExpress/clover #453: RLException: Unable to launch [sitl_0-3].

lucasmartins131, 2022-08-18:
After native installation with no errors reported I've got this message when running the roslaunch clover_simulation simulator.launch

RLException: Unable to launch [sitl_0-3]. 
If it is a script, you may be missing a '#!' declaration at the top.
The traceback for the exception was written to the log file

okalachev, 2022-08-23:
Can you please provide the full console log of your run? Without that it's hard to say what is the issue.

lucasmartins131, 2022-08-25:
[image]

okalachev, 2022-08-25:
Are you sure you sourced ROS' and `catkin_ws` `setup.bash` files?

lucasmartins131, 2022-08-25:
The bash file was wrong, but now I corrected it. One of the errors is not found anymore, Gazebo now is opening, but I am still getting the following errors. 

[image]

okalachev, 2022-08-25:
Have you symlinked the Clover airframe file to PX4 source?

```bash
ln -s ~/catkin_ws/src/clover/clover_simulation/airframes/* ~/PX4-Autopilot/ROMFS/px4fmu_common/init.d-posix/airframes/
```

lucasmartins131, 2022-08-25:
seams to be working better now. But I still got some errors: 

ERROR: cannot launch node of type [mavros_extras/visualization]: Cannot locate node of type [visualization] in package [mavros_extras]. Make sure file exists in package path and permission is set to executable (chmod +x)

[ERROR] [1661449371.634602863]: ros.mavros: Plugin ftp load exception: MultiLibraryClassLoader: Could not create object of class type mavros::std_plugins::FTPPlugin as no factory exists for it. Make sure that the library exists and was explicitly loaded through MultiLibraryClassLoader::loadLibrary()

lucasmartins131, 2022-08-25:
I also tried to call a service, with no sucesss: 

[image]

okalachev, 2022-08-25:
Are you sure you have installed mavros and mavros_extras correctly?

lucasmartins131, 2022-08-25:
I made the command rosdep install --from-paths src --ignore-src -y as mentioned in the "Native Setup". Also the mavros services are available when I call rosservice list. 
Is there other way to check the proper installation ?

okalachev, 2022-08-25:
Is there a connection between mavros and PX4? What `mavros/state` topic says? Please, add a log of your latest run, better in a text form as your screenshots don't cover the whole output.

lucasmartins131, 2022-08-25:
This is the full log after PX4 starts : 

px4 starting.

INFO  [px4] Calling startup script: /bin/sh etc/init.d-posix/rcS 0
Info: found model autostart file as SYS_AUTOSTART=4500
INFO  [param] selected parameter default file eeprom/parameters_4500
process[spawn_lenovo_29837_9067371672224030745-5]: started with pid [29895]
INFO  [parameters] BSON document size 444 bytes, decoded 444 bytes
[param] Loaded: eeprom/parameters_4500
process[mavros-6]: started with pid [29903]
process[rangefinder_frame-7]: started with pid [29912]
ERROR: cannot launch node of type [mavros_extras/visualization]: Cannot locate node of type [visualization] in package [mavros_extras]. Make sure file exists in package path and permission is set to executable (chmod +x)
process[web_video_server-9]: started with pid [29923]
process[vpe_publisher-10]: started with pid [29938]
process[optical_flow-11]: started with pid [29943]
process[map_flipped_frame-12]: started with pid [29956]
[ INFO] [1661453303.248368572]: ros.clover: using data topic
process[simple_offboard-13]: started with pid [29958]
process[main_camera_frame-14]: started with pid [29972]
[ INFO] [1661453303.298167409]: ros.nodelet: Loading nodelet /optical_flow of type clover/optical_flow to manager main_camera_nodelet_manager with the following remappings:
process[main_camera_nodelet_manager-15]: started with pid [29983]
[ INFO] [1661453303.313306207]: ros.nodelet: /camera_info -> /main_camera/camera_info
[ INFO] [1661453303.313362989]: ros.nodelet: /image_raw -> /main_camera/image_raw
process[main_camera/main_camera_markers-16]: started with pid [29995]
process[rosbridge_websocket-17]: started with pid [30003]
process[rosapi-18]: started with pid [30009]
[ INFO] [1661453303.377325281]: ros.roscpp: waitForService: Service [/main_camera_nodelet_manager/load_nodelet] has not been advertised, waiting...
process[tf2_web_republisher-19]: started with pid [30015]
process[led/led_effect-20]: started with pid [30029]
process[roswww_static-21]: started with pid [30041]
[ INFO] [1661453303.406398454]: ros.mavros: FCU URL: udp://@127.0.0.1:14580
[ INFO] [1661453303.416129740]: ros.rosconsole_bridge.mavconn: udp0: Bind address: 0.0.0.0:14555
[ INFO] [1661453303.416294218]: ros.rosconsole_bridge.mavconn: udp0: Remote address: 127.0.0.1:14580
[ INFO] [1661453303.417127223]: ros.mavros: GCS bridge disabled
[ INFO] [1661453303.465729877]: ros.clover: ready
INFO  [dataman] Unknown restart, data manager file './dataman' size is 11798680 bytes
PX4 SIM HOST: localhost
[ INFO] [1661453303.502866021]: ros.mavros: Plugin 3dr_radio blacklisted
[ INFO] [1661453303.502938657]: ros.mavros: Plugin actuator_control blacklisted
[ INFO] [1661453303.502959540]: ros.mavros: Plugin adsb blacklisted
INFO  [simulator] Waiting for simulator to accept connection on TCP port 4560
[ INFO] [1661453303.511222120]: ros.mavros: Plugin altitude loaded
[ INFO] [1661453303.541892229]: ros.mavros: Plugin altitude initialized
[ INFO] [1661453303.542022833]: ros.mavros: Plugin cam_imu_sync blacklisted
[ INFO] [1661453303.542145545]: ros.mavros: Plugin command loaded
[ INFO] [1661453303.544990766]: ros.nodelet: Initializing nodelet with 2 worker threads.
[ INFO] [1661453303.587524016]: ros.roscpp: waitForService: Service [/main_camera_nodelet_manager/load_nodelet] is now available.
[ INFO] [1661453303.588844168]: ros.roscpp: waitForService: Service [/led/set_leds] has not been advertised, waiting...
[ INFO] [1661453303.608839076]: ros.mavros: Plugin command initialized
[ INFO] [1661453303.609072907]: ros.mavros: Plugin companion_process_status blacklisted
[ INFO] [1661453303.609238083]: ros.mavros: Plugin debug_value blacklisted
[ INFO] [1661453303.611406745]: ros.mavros: Plugin distance_sensor loaded
[ INFO] [1661453303.673298504]: ros.mavros: Plugin distance_sensor initialized
[ INFO] [1661453303.673379451]: ros.mavros: Plugin fake_gps blacklisted
[ERROR] [1661453303.673703308]: ros.mavros: Plugin ftp load exception: MultiLibraryClassLoader: Could not create object of class type mavros::std_plugins::FTPPlugin as no factory exists for it. Make sure that the library exists and was explicitly loaded through MultiLibraryClassLoader::loadLibrary()
[ INFO] [1661453303.673821201]: ros.mavros: Plugin global_position loaded
[ INFO] [1661453303.732347626]: ros.mavros: Plugin global_position initialized
[ INFO] [1661453303.732469709]: ros.mavros: Plugin gps_rtk blacklisted
[ INFO] [1661453303.732525932]: ros.mavros: Plugin hil blacklisted
[ INFO] [1661453303.732575171]: ros.mavros: Plugin home_position blacklisted
[ INFO] [1661453303.732693204]: ros.mavros: Plugin imu loaded
[ INFO] [1661453303.762379937]: ros.mavros: Plugin imu initialized
[ INFO] [1661453303.762543857]: ros.mavros: Plugin landing_target blacklisted
[ INFO] [1661453303.762688010]: ros.mavros: Plugin local_position loaded
[ INFO] [1661453303.763001950]: ros.clover: ready
[ INFO] [1661453303.781296137]: ros.mavros: Plugin local_position initialized
[ INFO] [1661453303.781532761]: ros.mavros: Plugin log_transfer blacklisted
[ERROR] [1661453303.781941337]: ros.mavros: Plugin manual_control load exception: MultiLibraryClassLoader: Could not create object of class type mavros::std_plugins::ManualControlPlugin as no factory exists for it. Make sure that the library exists and was explicitly loaded through MultiLibraryClassLoader::loadLibrary()
[ INFO] [1661453303.782134100]: ros.mavros: Plugin mocap_pose_estimate blacklisted
[... 141 lines cut ...]
[... cut ...]
[ INFO] [1661453310.705098854, 4.560000000]: ros.mavros.imu: IMU: Attitude quaternion IMU detected!
INFO  [ecl/EKF] 4400000: EKF aligned, (rng hgt, IMU buf: 11, OBS buf: 8)
INFO  [ecl/EKF] 4404000: EKF aligned, (rng hgt, IMU buf: 11, OBS buf: 8)
INFO  [ecl/EKF] 4408000: EKF aligned, (rng hgt, IMU buf: 11, OBS buf: 8)
INFO  [ecl/EKF] 4408000: EKF aligned, (rng hgt, IMU buf: 11, OBS buf: 8)
[ INFO] [1661453316.155726710, 10.000000000]: ros.clover: publish zero
[ INFO] [1661453326.176704818, 20.000000000]: ros.clover: publish zero

lucasmartins131, 2022-08-25:
This is the output of mavros/state: 

header: 
  seq: 136
  stamp: 
    secs: 136
    nsecs: 380000000
  frame_id: ''
connected: True
armed: False
guided: False
manual_input: True
mode: "MANUAL"
system_status: 3

okalachev, 2022-08-25:
I don't see any errors here. The `"State timeout, check mavros settings"` could happen, only when mavros is not connected (`connected: False` in mavros state).

lucasmartins131, 2022-08-25:
What about this: 

ERROR: cannot launch node of type [mavros_extras/visualization]: Cannot locate node of type [visualization] in package [mavros_extras]. Make sure file exists in package path and permission is set to executable (chmod +x)

[ERROR] [1661453303.673703308]: ros.mavros: Plugin ftp load exception: MultiLibraryClassLoader: Could not create object of class type mavros::std_plugins::FTPPlugin as no factory exists for it. Make sure that the library exists and was explicitly loaded through MultiLibraryClassLoader::loadLibrary()

lucasmartins131, 2022-08-25:
And Also this behavior when I try to call the navigate service : 

lucas@lenovo:~$ rosservice call /navigate "{x: 0.0, y: 0.0, z: 5.0, yaw: 0.0, yaw_rate: 0.0, speed: 0.0, frame_id: 'body', auto_arm: true}" 
success: False
message: "State timeout, check mavros settings"

okalachev, 2022-08-25:
The first one makes it impossible to monitor the drone in rviz using a quadcopter-like model, in `vehicle_markers` topic, but it doesn't affect the flight. I would suggest to reinstall the mavros_extras package.

The second one, seemly, is error loading some of (not required) mavros plugin, but you say, that the rest of mavros is working, so it's ok.

okalachev, 2022-08-25:
> And Also this behavior when I try to call the navigate service :
> 
> lucas@lenovo:~$ rosservice call /navigate "{x: 0.0, y: 0.0, z: 5.0, yaw: 0.0, yaw_rate: 0.0, speed: 0.0, frame_id: 'body', auto_arm: true}"
> success: False
> message: "State timeout, check mavros settings"

So the topic `mavros/state` consist connected: True, and in the same time you are getting this error? This is very weird, like you're using different rosmasters or something.

lucasmartins131, 2022-08-25:
> > And Also this behavior when I try to call the navigate service :
> > lucas@lenovo:~$ rosservice call /navigate "{x: 0.0, y: 0.0, z: 5.0, yaw: 0.0, yaw_rate: 0.0, speed: 0.0, frame_id: 'body', auto_arm: true}"
> > success: False
> > message: "State timeout, check mavros settings"
> 
> So the topic `mavros/state` consists `connected: True`, and in the same time you are getting this error? This is very weird, like you're using different rosmasters or something.

Yes it is showing connected: True. But calling this service is not working)

lucasmartins131, 2022-08-25:
> So the topic `mavros/state` consists `connected: True`, and in the same time you are getting this error? This is very weird, like you're using different rosmasters or something.

> The first one makes it impossible to monitor the drone in rviz using a quadcopter-like model, in `vehicle_markers` topic, but it doesn't affect the flight. I would suggest to reinstall the [...]

Do you know how to reeinstall mavros without calling via rosdep ?

okalachev, 2022-08-25:
> Do you know how to reeinstall mavros without calling via rosdep ?

`sudo apt-get install ros-noetic-mavros`

okalachev, 2022-08-25:
> Yes it is showing connected: True. But calling this service is not working)

Are you sure about that? Can you please attach a screenshot where you echo the mavros/state topic and then call the service getting this error. Because the only condition for that error is `connected: False` in that topic.

lucasmartins131, 2022-08-25:
> Are you sure about that? Can you please attach a screenshot where you echo the `mavros/state` topic and then call the service getting this error. Because the only condition for that error is [...]

[image]

okalachev, 2022-08-25:
OK, I know what is happening. I've mistaken state timeout error with no connection error. mavros publishes connected state, but then falls down due to Plugin ftp load exception. 

You can either try to fix mavros reinstalling it, or just disable ftp plugin removing this line here https://github.com/CopterExpress/clover/blob/master/clover/launch/mavros.launch#L48. This file is located in `~/catkin_ws/src/clover/launch`.

lucasmartins131, 2022-08-25:
> OK, I know what is happening. I've mistaken state timeout error with no connection error. mavros publishes connected state, but then, seemingly, falls down due to Plugin ftp load exception, causing state timeout error.
> 
> You can either try to fix mavros reinstalling it, or just disable ftp plugin removing this line here https://github.com/CopterExpress/clover/blob/master/clover/launch/mavros.launch#L48. This file is located in `~/catkin_ws/src/clover/launch`.

Removed the FTP line from the launch file, and tried to launch it again. No sucess. Do I have to compile the folder again ?

okalachev, 2022-08-25:
No, you don't. Does mavros run well now without that ftp error? Is the /mavros/state topic published constantly at rate 1 Hz with connected: true?

lucasmartins131, 2022-08-25:
This is the output of rostopic hz /mavros/state 

[image]

lucasmartins131, 2022-08-25:
> No, you don't. Does mavros run well now without that ftp error? Is the /mavros/state topic published constantly at rate 1 Hz with connected: true?

No improvements after removing the ftp from the launch file

closed 2023-03-06 by okalachev