CopterExpress/clover #454: catkin make fail due to missing files error

AlexWUrobot, 2022-08-18:
Thank you very much for your help
I have installed MAVROS and AutoPilot and QGroundControl successfully, but 
when I try to catkin_make the clover_ws.
It shows that No such file or directory 
[image]

okalachev, 2022-08-18:
Seems as mavros wasn't correctly installed, because its dependent package `mavros_msgs` cannot be found. Or maybe you haven't sourced ROS' `setup.bash`?

AlexWUrobot, 2022-08-18:
Yes, at the beginning, I put mavros in catkin_ws and catkin_build successfully and source devel/setup.bash 
[image]
Then, it still cannot find the <mavros_msg/State.h>
so I git clone one more mavros in clover_ws, and then catkin build again
[image]

However, it shows the clover file is abandon from catkin build
[image]
[image]
Next, I think that mavros already is caktin build successfully 
so I try to catkin build the clover package by this command "catkin build -j1 -i -v --no-deps clover"
[image]
However, it shows the same error "mavros_msg/State.h" in the end 
[image]

Thanks again for your reply : )

okalachev, 2022-08-18:
First, ROS Noetic is recommended to use with Clover now (not Melodic). Second, have you followed this tutorial https://clover.coex.tech/en/simulation_native.html ?

AlexWUrobot, 2022-08-18:
Thank you very much.
Yes, I am using Melodic on Jetson TX1 with Ubuntu 18.04.
Is it feasible to use clover package on Melodic?

If not, do you know any other package can support speed and position control at the same time like this below line ?  
navigate(x=3, y=0, z=0, speed=0.5, yaw=float('nan'), yaw_rate=0.2, frame_id='body')

Because PX4-AutoPilot only support Position control sample, I am frustrated to control the speed and position at the same time
https://github.com/PX4/PX4-Autopilot/blob/main/integrationtests/python_src/px4_it/mavros/mavros_offboard_posctl_test.py

Really appreciate again

AlexWUrobot, 2022-08-18:
I tried to remove mavros package from clover_ws and keep mavros package in catkin_ws.
Next, I run " sudo apt-get install ros-melodic-mavros*"
Then, catkin_make the clover_ws and the error change to this 

[image]

AlexWUrobot, 2022-08-19:
It looks 100% catkin build, so I ran roslaunch px4 mavros_posix_sitl.launch and python flight.py
roslaunch px4 mavros_posix_sitl.launch (works, publish all /mavros/ topics data)
but python flight.py shows error "rosgraph.masterapi.MasterError: no provider"
[image]

okalachev, 2022-08-19:
Are you trying to run the simulation on Jetson, or the Clover packages itself to fly the real drone?

AlexWUrobot, 2022-08-19:
No, I run the simulation and the Clover Packages on a PC with Ubuntu 18.04  
Next step
Yes, once the Clover Package goes well in PC simulation, I will run the Clover package on Jetson to offboard control a real drone.
Thanks for your patience and help again : )

okalachev, 2022-08-19:
To run clover services, you need to run `clover.launch` file. And you don't need `px4` package at all, because PX4 runs on a flight controller.

AlexWUrobot, 2022-08-19:
[image]
[image]
If do not run px4, there is an error showing like this.

okalachev, 2022-08-19:
If you do not have a vl53l1x rangefinder and a ws281x LED strip, you can turn their drivers off in `clover.launch`.

closed 2023-03-06 by okalachev