CopterExpress/clover #484: catkin_make [Errno 13] Permission denied: '/home/user/catkin_ws/build'

shourez, 2023-03-05:
My goal is to use block programming for simulation. I tried to do it using VM under Windows 10. On my AMD Athlon(tm) II X4 635 Processor, it failed. 

Then I tried to install clover simulation following https://clover.coex.tech/en/simulation_native.html. After the first catkin_make command, I receive the error mentioned above. 

mkdir -p ~/catkin_ws/src
cd ~/catkin_ws
catkin_make

Of course after these commands I had no more than an empty /catkin_ws/src folder. What was catkin_make supposed to do? Building from what?

okalachev, 2023-03-05:
>On my AMD Athlon(tm) II X4 635 Processor, it failed.

Actually, it should not fail, what do you mean by failing?

>After the first catkin_make command, I receive the error mentioned above.

Looks like some kind of permissions issue, are you sure you logged in as the user `user`, or maybe some else? Haven't you created `catkin_ws` directory from `root` maybe?

>Of course after these commands I had no more than an empty /catkin_ws/src folder. What was catkin_make supposed to do? Building from what?

This command is supposed to create a new Catkin workspace, including `devel/setup.bash` file. You can refer to the official documentation if it's not clear: http://wiki.ros.org/catkin/Tutorials/create_a_workspace.

shourez, 2023-03-06:
Yes, I did exactly that mistake, creating catkin_ws directory from root. Thank you for pointing it out. I suppose it is a common issue, perhaps the manual could contain a warning about it. 

Now after the step roslaunch clover_simulation simulator.launch Gazebo crushed, this is terminal output 

[Msg] Loading world file [/home/malyzhenkov/catkin_ws/src/clover/clover_simulation/resources/worlds/clover_aruco.world]
[ INFO] [1678130989.683803140]: waitForService: Service [/gazebo/set_physics_properties] is now available.
[ INFO] [1678130989.823727223, 0.048000000]: Physics dynamic reconfigure ready.
[INFO] [1678130989.831394, 0.056000]: Calling service /gazebo/spawn_urdf_model
Service call failed: transport error completing service call: unable to receive data from sender, check sender's logs for details
[ERROR] [1678131002.943261, 0.216000]: Spawn service failed. Exiting.
[Err] [Connection.cc:547] Connection[17] Closed during Read
[Err] [TransportIface.cc:385] Unable to read from master
[Err] [GLWidget.cc:900] Unable to connect to a running Gazebo master.
Illegal instruction (core dumped)
[gazebo-2] process has died [pid 23231, exit code 132, cmd /opt/ros/noetic/lib/gazebo_ros/gzserver --verbose -e ode /home/malyzhenkov/catkin_ws/src/clover/clover_simulation/resources/worlds/clover_aruco.world __name:=gazebo __log:=/home/malyzhenkov/.ros/log/3edf9c1c-bc55-11ed-a75b-47f8146f0bc7/gazebo-2.log].
log file: /home/malyzhenkov/.ros/log/3edf9c1c-bc55-11ed-a75b-47f8146f0bc7/gazebo-2*.log
[Err] [Visual.cc:383] Service name [/shininess] not advertised, not attempting to load shininess for visual with name [clover::led_link::led_link_visual].
[spawn_malyzhenkov_System_Product_Name_23201_6778556927274380805-5] process has died [pid 23324, exit code 1, cmd /opt/ros/noetic/lib/gazebo_ros/spawn_model -urdf -param drone_description -model clover -z 0.3 __name:=spawn_malyzhenkov_System_Product_Name_23201_6778556927274380805 __log:=/home/malyzhenkov/.ros/log/3edf9c1c-bc55-11ed-a75b-47f8146f0bc7/spawn_malyzhenkov_System_Product_Name_23201_6778556927274380805-5.log].
log file: /home/malyzhenkov/.ros/log/3edf9c1c-bc55-11ed-a75b-47f8146f0bc7/spawn_malyzhenkov_System_Product_Name_23201_6778556927274380805-5*.log
[ INFO] [1678131003.763726962, 0.216000000]: LedController has started (as client) in namespace ''
[Err] [Visual.cc:383] Service name [/shininess] not advertised, not attempting to load shininess for visual with name [clover::led_link::led_link_visual_1].
[Err] [Visual.cc:383] Service name [/shininess] not advertised, not attempting to load shininess for visual with name [clover::led_link::led_link_visual_2].
[Err] [Visual.cc:383] Service name [/shininess] not advertised, not attempting to load shininess for visual with name [clover::led_link::led_link_visual_3].
[Err] [Visual.cc:383] Service name [/shininess] not advertised, not attempting to load shininess for visual with name [clover::led_link::led_link_visual_4].
[Err] [Visual.cc:383] Service name [/shininess] not advertised, not attempting to load shininess for visual with name [clover::led_link::led_link_visual_5].
[Err] [Visual.cc:383] Service name [/shininess] not advertised, not attempting to load shininess for visual with name [clover::led_link::led_link_visual_6].
[Err] [Visual.cc:383] Service name [/shininess] not advertised, not attempting to load shininess for visual with name [clover::led_link::led_link_visual_7].
[Err] [Visual.cc:383] Service name [/shininess] not advertised, not attempting to load shininess for visual with name [clover::led_link::led_link_visual_8].
[Err] [Visual.cc:383] Service name [/shininess] not advertised, not attempting to load shininess for visual with name [clover::led_link::led_link_visual_9].
[Err] [Visual.cc:383] Service name [/shininess] not advertised, not attempting to load shininess for visual with name [clover::led_link::led_link_visual_10].
[Err] [Visual.cc:383] Service name [/shininess] not advertised, not attempting to load shininess for visual with name [clover::led_link::led_link_visual_11].
[Err] [Visual.cc:383] Service name [/shininess] not advertised, not attempting to load shininess for visual with name [clover::led_link::led_link_visual_13].
[Err] [Visual.cc:383] Service name [/shininess] not advertised, not attempting to load shininess for visual with name [clover::led_link::led_link_visual_12].
[Err] [Visual.cc:383] Service name [/shininess] not advertised, not attempting to load shininess for visual with name [clover::led_link::led_link_visual_14].
[Err] [Visual.cc:383] Service name [/shininess] not advertised, not attempting to load shininess for visual with name [clover::led_link::led_link_visual_15].
[Err] [Visual.cc:383] Service name [/shininess] not advertised, not attempting to load shininess for visual with name [clover::led_link::led_link_visual_16].
[Err] [Visual.cc:383] Service name [/shininess] not advertised, not attempting to load shininess for visual with name [clover::led_link::led_link_visual_18].
[Err] [Visual.cc:383] Service name [/shininess] not advertised, not attempting to load shininess for visual with name [clover::led_link::led_link_visual_17].
[Err] [Visual.cc:383] Service name [/shininess] not advertised, not attempting to load shininess for visual with name [clover::led_link::led_link_visual_19].
[Err] [Visual.cc:383] Service name [/shininess] not advertised, not attempting to load shininess for visual with name [clover::led_link::led_link_visual_20].
[... cut ...]
[Wrn] [ColladaLoader.cc:1928] Triangle input semantic: 'COLOR' is currently not supported
[Wrn] [ColladaLoader.cc:1928] Triangle input semantic: 'COLOR' is currently not supported
[Wrn] [ColladaLoader.cc:1928] Triangle input semantic: 'COLOR' is currently not supported
[Wrn] [Publisher.cc:135] Queue limit reached for topic /gazebo/default/user_camera/pose, deleting message. This warning is printed only once.
 
libcurl: (6) Could not resolve host: fuel.gazebosim.org
 
libcurl: (6) Could not resolve host: fuel.ignitionrobotics.org

okalachev, 2023-03-07:
>Yes, I did exactly that mistake, creating catkin_ws directory from root. Thank you for pointing it out. I suppose it is a common issue, perhaps the manual could contain a warning about it.

We'll consider that.

>Now after the step roslaunch clover_simulation simulator.launch Gazebo crushed, this is terminal output

>```Illegal instruction (core dumped)```

I see this issue for the first time, really unobvious problem. Maybe there was some issues or errors during the installation steps?

Can just run Gazebo (`gazebo`)? What is Gazebo version installed? Maybe try to update?

okalachev, 2023-03-08:
Does any 3D graphics app work? Maybe you haven't enabled 3D acceleration?

shourez, 2023-03-10:
> Does any 3D graphics app work? Maybe you haven't enabled 3D acceleration?

I installed and used Blender, it works fine.

okalachev, 2023-03-23:
It's actually not obvious for me what causes the error. It's something related to virtualization on this machine.

Sorry for a long answer. You might get quicker answer using official support Telegram chat: https://t.me/COEXHelpdesk.

shourez, 2023-03-27:
> It's actually not obvious for me what causes the error. It's something related to virtualization on this machine.
> 
> Sorry for a long answer. You might get quicker answers using official support Telegram chat: https://t.me/COEXHelpdesk.

I thought virtualization was supposed to be hardware-independent. Now I see it far from that.

closed 2023-06-13 by okalachev