CopterExpress/clover #485: Error: No connection to FCU

shourez, 2023-03-05:
The same PC as in https://github.com/CopterExpress/clover/issues/484.

In VM, block programming interface doesn't connect to Gazebo
[image]

okalachev, 2023-03-05:
Can you please provide the full launch command output, from the terminal?

shourez, 2023-03-06:
Here it is

[Msg] Waiting for master.
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 192.168.238.129
[ INFO] [1678127390.344030690]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1678127390.351515672]: waitForService: Service [/gazebo_gui/set_physics_properties] has not been advertised, waiting...
[Msg] Waiting for master.
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 192.168.238.129
[Msg] Loading world file [/home/clover/catkin_ws/src/clover/clover_simulation/resources/worlds/clover_aruco.world]
[ INFO] [1678127391.916267973]: waitForService: Service [/gazebo/set_physics_properties] is now available.
[INFO] [1678127391.973745, 0.000000]: wait_for_service(/gazebo/spawn_urdf_model): finally were able to contact [rosrpc://clover-dev:35257]
[INFO] [1678127391.986251, 0.000000]: Calling service /gazebo/spawn_urdf_model
[INFO] [1678127392.989377, 0.004000]: Spawn status: SpawnModel: Successfully spawned entity
[ INFO] [1678127393.105277662, 0.004000000]: Physics dynamic reconfigure ready.
[spawn_clover_dev_3051_7709871114269057712-4] process has finished cleanly
log file: /home/clover/.ros/log/382a859e-bc4b-11ed-a7bf-93a94191dd8e/spawn_clover_dev_3051_7709871114269057712-4*.log
[ INFO] [1678127394.532303295, 0.004000000]: Camera Plugin: Using the 'robotNamespace' param: '/'
[ INFO] [1678127394.567942426, 0.004000000]: Camera Plugin (ns = /)  <tf_prefix_>, set to ""
[Msg] Connecting to PX4 SITL using TCP
[Msg] Lockstep is enabled
[Msg] Speed factor set to: 1
[ INFO] [1678127395.738227341, 0.004000000]: LedController has started (as server) in namespace ''
[Wrn] [gazebo_mavlink_interface.cpp:152] No identifier on joint. Using 0 as default sensor ID
[Msg] Using MAVLink protocol v2.0
INFO  [simulator] Simulator connected on TCP port 4560.
[ INFO] [1678127395.983872749, 0.004000000]: Initialize LED Controller
[ INFO] [1678127396.007141743, 0.008000000]: ros.roscpp: waitForService: Service [/led/set_leds] is now available.
[ INFO] [1678127396.261686602, 0.044000000]: ros.clover: ready
INFO  [commander] LED: open /dev/led0 failed (22)
* PWM_AUX_OUT: curr: 0 -> new: 1234
* PWM_MAIN_OUT: curr: 0 -> new: 1234
Illegal instruction (core dumped)
[gazebo-1] process has died [pid 3068, exit code 132, cmd /opt/ros/noetic/lib/gazebo_ros/gzserver --verbose -e ode /home/clover/catkin_ws/src/clover/clover_simulation/resources/worlds/clover_aruco.world __name:=gazebo __log:=/home/clover/.ros/log/382a859e-bc4b-11ed-a7bf-93a94191dd8e/gazebo-1.log].
log file: /home/clover/.ros/log/382a859e-bc4b-11ed-a7bf-93a94191dd8e/gazebo-1*.log
[Err] [Visual.cc:410] Service call [/shininess] timed out
[Err] [Visual.cc:383] Service name [/shininess] not advertised, not attempting to load shininess for visual with name [clover::led_link::led_link_visual].
[ INFO] [1678127401.923273320, 0.084000000]: LedController has started (as client) in namespace ''
[Err] [Visual.cc:383] Service name [/shininess] not advertised, not attempting to load shininess for visual with name [clover::led_link::led_link_visual_1].
[Err] [Visual.cc:383] Service name [/shininess] not advertised, not attempting to load shininess for visual with name [clover::led_link::led_link_visual_2].
[Err] [Visual.cc:383] Service name [/shininess] not advertised, not attempting to load shininess for visual with name [clover::led_link::led_link_visual_3].
[Err] [Visual.cc:383] Service name [/shininess] not advertised, not attempting to load shininess for visual with name [clover::led_link::led_link_visual_4].
[Err] [Visual.cc:383] Service name [/shininess] not advertised, not attempting to load shininess for visual with name [clover::led_link::led_link_visual_5].
[Err] [Visual.cc:383] Service name [/shininess] not advertised, not attempting to load shininess for visual with name [clover::led_link::led_link_visual_6].
[Err] [Visual.cc:383] Service name [/shininess] not advertised, not attempting to load shininess for visual with name [clover::led_link::led_link_visual_7].
[Err] [Visual.cc:383] Service name [/shininess] not advertised, not attempting to load shininess for visual with name [clover::led_link::led_link_visual_8].
[Err] [Visual.cc:383] Service name [/shininess] not advertised, not attempting to load shininess for visual with name [clover::led_link::led_link_visual_9].
[Err] [Visual.cc:383] Service name [/shininess] not advertised, not attempting to load shininess for visual with name [clover::led_link::led_link_visual_10].
[Err] [Visual.cc:383] Service name [/shininess] not advertised, not attempting to load shininess for visual with name [clover::led_link::led_link_visual_11].
[Err] [Visual.cc:383] Service name [/shininess] not advertised, not attempting to load shininess for visual with name [clover::led_link::led_link_visual_12].
[Err] [Visual.cc:383] Service name [/shininess] not advertised, not attempting to load shininess for visual with name [clover::led_link::led_link_visual_13].
[Err] [Visual.cc:383] Service name [/shininess] not advertised, not attempting to load shininess for visual with name [clover::led_link::led_link_visual_14].
[Err] [Visual.cc:383] Service name [/shininess] not advertised, not attempting to load shininess for visual with name [clover::led_link::led_link_visual_15].
[Err] [Visual.cc:383] Service name [/shininess] not advertised, not attempting to load shininess for visual with name [clover::led_link::led_link_visual_16].
[... cut ...]
[ INFO] [1678127449.932253728, 0.092000000]: ros.clover: State timeout, check mavros settings
[ERROR] [1678127449.935620, 0.092000]: rosout: State timeout, check mavros settings
Traceback (most recent call last):
  File "/home/clover/catkin_ws/src/clover/clover_blocks/src/clover_blocks", line 115, in program_thread
    exec(req.code, g)
  File "<string>", line 31, in <module>
  File "<string>", line 16, in navigate_wait
Exception: State timeout, check mavros settings
[INFO] [1678127449.951451, 0.092000]: rosout: Program terminated

okalachev, 2023-03-07:
So you got the same errors in the [VM image](https://clover.coex.tech/en/simulation_vm.html) and after the [manual installation](https://clover.coex.tech/en/simulation_native.html)?

shourez, 2023-03-08:
> So you got the same errors in the [VM image](https://clover.coex.tech/en/simulation_vm.html) and after the [manual installation](https://clover.coex.tech/en/simulation_native.html)?

Looks so. But VM at least starts Gazebo, while manual installation doesn't do even that. It may be somehow connected to hardware.

okalachev, 2025-01-10:
Closed as stale.

closed 2025-01-10 by okalachev