CopterExpress/clover #487: Error: name 'set_yaw' is not defined

shourez, 2023-03-27:
Attempting to yaw a copter with block programming fails with this error.

Terminal output:

ent 0] Subscribed to /clover_blocks/prompt
[INFO] [1679918060.244303, 123.360000]: rosout: Run program
[ INFO] [1679918060.317255022, 123.424000000]: ros.clover: switch to OFFBOARD
[ WARN] [1679918060.327299246, 123.428000000]: ros.mavros.cmd: CMD: Unexpected command 176, result 0
[ INFO] [1679918061.344813972, 124.436000000]: ros.clover: arming
INFO  [commander] Armed by external command	
INFO  [tone_alarm] arming warning
INFO  [commander] Takeoff detected	
[ERROR] [1679918068.696910, 131.784000]: rosout: name 'set_yaw' is not defined
Traceback (most recent call last):
  File "/home/clover/catkin_ws/src/clover/clover_blocks/src/clover_blocks", line 115, in program_thread
    exec(req.code, g)
  File "<string>", line 40, in <module>
NameError: name 'set_yaw' is not defined
[INFO] [1679918068.698212, 131.784000]: rosout: Program terminated
[ INFO] [1679918068.699245091, 131.784000000]: ros.clover: led: notify error

Screenshot

[image]

shourez, 2023-03-28:
> I suppose you're using the virtual machine image version 1.4, which is a pre-release and is completely tested.
> 
> But this is a bug, thanks for the report. It is related to autonomous flight API redisign and introducing the new `set_yaw` ROS service.
> 
> I believe I fixed it in commit [ae05710](https://github.com/CopterExpress/clover/commit/ae05710a3768151f3d16d5cd4089652fac7f77f7), so if you can apply it into your workspace and test, please, report to us the results, so we could close the issue.

The image file was called "clover-devel_v1.4-alpha.1.ova".

I don't understand how exactly I should apple this commit.

I pasted  

		if (rosDefinitions.setYaw) {
			code += `set_yaw = rospy.ServiceProxy('set_yaw', srv.SetYaw)\n`;
		}

into ~/clover/catkin_ws/src/clover/clover_blocks/www/python.js. Nothing changed.

okalachev, 2023-03-28:
>The image file was called "clover-devel_v1.4-alpha.1.ova".

All the images, that contain `v1.4` in their name are pre-release, we haven't released the version 1.4 yet.

>I don't understand how exactly I should apple this commit.

You can apply the commit using git, or manually, then you should refresh the page in web browser dropping all the cache (Ctrl+Shift+R in Firefox).

shourez, 2023-03-28:
> All the images, that contain `v1.4` in their name are pre-release, we haven't released the version 1.4 yet.
>
>
> You can apply the commit using git, or manually, then you should refresh the page in web [...]

I don't know how to apply commits through. I manually pasted those lines in that file, that's all. Ctrl+Shift+R in the browser helped.