CopterExpress/clover #489: Error with Hardware Implementation

Prathap20, 2023-06-01:
Hello! 

I have been trying to implement the Clover package for my drone. I was successful in replicating the simulation part. But now I am having problems with the hardware implementation. The following are my device descriptions:

1. Holybro s500 V2 drone kit
2. PX4 Cube pilot - Orange Cube
3. Raspberry Pi 4 + Raspberry Pi camera module

I have setup the clover rpi image on the pi and was able to compile it properly. The web tools seems to work and the aruco detection is working fine.

The problem is with the FCU and Qgroundcontrol part. I have set the PX4 parameters as stated in the Clover documentation(I know the parameters for my drone is different but wanted to check it out anyway).

When I run the automated selfcheck I am getting a lot of errors. I have attached the shell script below.

Kindly help and guide me on what I should do differently so that I can finish this project successfully .

Thank you!

pi@clover-3808:~ $ rosrun clover selfcheck.py
[INFO]: Performing selfcheck...
[INFO]: Image: version: v0.23
[INFO]: Board: Raspberry Pi 4 Model B Rev 1.4
[WARN]: clover.service: WARN: ros.rosconsole_bridge.mavconn: udp1: resolve error: Host not found (non-authoritative), try again later
[WARN]: clover.service: FATAL: ros.mavros: GCS: DeviceError:udp: resolve:Bind address resolve failed
[WARN]: clover.service: WARN: ros.mavros.sys: VER: broadcast request timeout, retries left 4
[WARN]: clover.service: WARN: ros.mavros.sys: VER: broadcast request timeout, retries left 3
[WARN]: clover.service: WARN: ros.mavros.cmd: CMD: Command 520 -- wait ack timeout (4)
[WARN]: clover.service: WARN: ros.mavros.sys: VER: unicast request timeout, retries left 2
[WARN]: clover.service: WARN: ros.mavros.param: PR: request list timeout, retries left 2
[WARN]: clover.service: WARN: ros.mavros.param: PR: request list timeout, retries left 1
[WARN]: clover.service: WARN: ros.mavros.param: PR: request list timeout, retries left 0
[WARN]: clover.service: WARN: ros.mavros.sys: VER: unicast request timeout, retries left 1
[WARN]: clover.service: WARN: ros.mavros.sys: VER: unicast request timeout, retries left 0
[WARN]: clover.service: WARN: ros.mavros.sys: VER: your FCU don't support AUTOPILOT_VERSION, switched to default capabilities
[INFO]: Network: OK
[INFO]: FCU: no version data available from SITL
[INFO]: FCU: not Clover PX4 firmware, check https://clover.coex.tech/firmware
[WARN]: FCU: unable to retrieve PX4 parameter SYS_MC_EST_GROUP
[WARN]: FCU: unknown selected estimator: None
[WARN]: FCU: unable to retrieve PX4 parameter SENS_BOARD_ROT
[WARN]: FCU: unable to retrieve PX4 parameter CBRK_USB_CHK
[WARN]: FCU: set parameter CBRK_USB_CHK to 197848 for flying with USB connected
[WARN]: FCU: incorrect cell voltage: 6.53 V, https://clover.coex.tech/power
[INFO]: IMU: OK
[WARN]: Local position: no local position
[WARN]: Velocity estimation: no velocity estimation
[INFO]: Global position (GPS): no global position
[WARN]: Preflight status: no data from FCU
[INFO]: Main camera: camera is oriented downward, cable from camera goes backward
[INFO]: ArUco markers: aruco_detect/length = 0.33 m
[INFO]: ArUco markers: aruco_detector/known_tilt = map (ALL markers are on the floor)
[INFO]: ArUco markers: aruco_map/known_tilt = 
[WARN]: ArUco markers: cannot read aruco_map/visualization topic
[WARN]: ArUco markers: no map detection
[INFO]: Simple offboard node: OK
[WARN]: Optical flow: no optical flow data (from Raspberry)
[WARN]: Vision position estimate: no VPE or MoCap messages
[WARN]: Rangefinder: no rangefinder data from Raspberry
[WARN]: Rangefinder: no rangefinder data from PX4
[INFO]: RPi health: OK
[INFO]: CPU usage: OK
[INFO]: Boot duration: OK

okalachev, 2023-06-01:
By "implement" you mean to run, I guess?

What firmware version do you run on PX4 Cube pilot?

I'm not sure also, is there is connection between RPi and the flight controller? What do you see in `/mavros/state` topic for a start (`rostopic echo /mavros/state`)?

okalachev, 2023-06-01:
mavros doesn't seem to be connected to the FC (although is received IMU messages based on the posted selfcheck report, which is weird).

Anyway, Clover’s packages by default connect to `/dev/px4fmu` device file, which is created by udev only for certain flight controller models listed here https://github.com/CopterExpress/clover/blob/master/clover/udev/99-px4fmu.rules.

Did this file appear on your system? if not, you can add a rule to udev file, or just change the `usb_device` argument in `mavros.launch` file to existing usb device file, namely, to `/dev/ttyACM0`.

Prathap20, 2023-06-02:
Hello Mr. Kalachev! 

I changed the usb_device argument to /dev/ttyACM0. The FCU seems to be connected to the RPi. 

But still I am not getting the **Local position** and **velocity estimation**. I have studied in the documentation that only if the velocity estimation passes, simple offboard should be performed. 

How do I fix this problem? Kindly help me with this issue.

Thank you!

Selfcheck output:

pi@clover-3808:~ $ rosrun clover selfcheck.py 
[INFO]: Performing selfcheck...
[INFO]: Image: version: v0.23
[INFO]: Board: Raspberry Pi 4 Model B Rev 1.4
[INFO]: clover.service: OK
[INFO]: Network: OK
[INFO]: FCU: CUBEPILOT_CUBEORANGE
[INFO]: FCU: Release 1.13.3 (17630207)
[INFO]: FCU: not Clover PX4 firmware, check https://clover.coex.tech/firmware
[WARN]: FCU: unable to retrieve PX4 parameter SYS_MC_EST_GROUP
[WARN]: FCU: unknown selected estimator: None
[WARN]: FCU: unable to retrieve PX4 parameter SENS_BOARD_ROT
[WARN]: FCU: unable to retrieve PX4 parameter CBRK_USB_CHK
[WARN]: FCU: set parameter CBRK_USB_CHK to 197848 for flying with USB connected
[WARN]: FCU: incorrect cell voltage: 6.19 V, https://clover.coex.tech/power
[INFO]: IMU: OK
[WARN]: Local position: no local position
[WARN]: Velocity estimation: no velocity estimation
[INFO]: Global position (GPS): no global position
[INFO]: Preflight status: OK
[INFO]: Main camera: camera is oriented downward, cable from camera goes backward
[INFO]: ArUco markers: aruco_detect/length = 0.33 m
[INFO]: ArUco markers: aruco_detector/known_tilt = map (ALL markers are on the floor)
[INFO]: ArUco markers: aruco_map/known_tilt = 
[WARN]: ArUco markers: cannot read aruco_map/visualization topic
[WARN]: ArUco markers: no map detection
[INFO]: Simple offboard node: OK
[WARN]: Optical flow: no optical flow data (from Raspberry)
[WARN]: Vision position estimate: unable to retrieve PX4 parameter SYS_MC_EST_GROUP
[WARN]: Rangefinder: no rangefinder data from Raspberry
[WARN]: Rangefinder: no rangefinder data from PX4
[INFO]: RPi health: OK
[INFO]: CPU usage: OK
[INFO]: Boot duration: OK

okalachev, 2023-06-07:
I guess we've continued in the support chat, so I can close this?

okalachev, 2023-06-13:
Continued in support chat.

closed 2023-06-13 by okalachev

Prathap20, 2023-06-14:
Hello Mr. Kalachev!

Sorry for the late reply. I was on vacation sonI couldn't reply.

Yes you can close the issue.

Thank you very much for your support.

Best regards,
Prathap.