CopterExpress/clover #490: Jetson nano clover drone

youngjae41, 2023-06-07:
Hello

I am using Cloverdrone well.

Can I use the clover code on the jetson nano and not on the RPI?

okalachev, 2023-06-07:
Yes, but you to need to compile all the Clover packages yourself. There is a manual on that: https://clover.coex.tech/en/jetson_nano.html, but note, that it's a bit outdated. For example, you need to use Ubuntu 20.04 instead of Ubuntu 18.04, `noetic` instead of `melodic`, OpenCV 4.2 instead of OpenCV 3.2

Feel free to report if you have succeed to compile on a Jetson Nano, or if there are some issues.

youngjae41, 2023-06-08:
@okalachev Thank you for answer. All builds were successful. opencv was installed with version 4.2, which is automatically installed when installing noetic, and there seems to be no problem. One question is, does it work the same even if I use a jetson dedicated camera?

okalachev, 2023-06-08:
It should work with another camera but it's better to perform the [camera calibration](https://clover.coex.tech/en/camera_calibration.html), since the camera calibration parameters by default are for RPi's fisheye camera.

youngjae41, 2023-06-09:
@okalachev 
Thank you for a detailed description.

closed 2023-06-09 by youngjae41