CopterExpress/clover #491: Observatory simulation px4 1.13 or later version operation error

youngjae41, 2023-06-09:
Hello I am testing on PX4 latest firmware. It works with some modifications, but I get the following error

WARN [health_and_arming_checks] Preflight Fail: Attitude failure (roll)
WARN [health_and_arming_checks] Preflight Fail: heading estimate not stable
WARN [health_and_arming_checks] Preflight Fail: height estimate error
WARN [health_and_arming_checks] Preflight Fail: Yaw estimate error
WARN [health_and_arming_checks] Preflight Fail: Attitude failure (roll)

okalachev, 2023-06-09:
We have a beta version of virtual machine image with the simulation built with PX4 v1.13.3, and it seems to work: https://github.com/CopterExpress/clover_vm/releases/tag/v1.4.

So no problem should be with this version, if you install all the software according to the manual.

youngjae41, 2023-06-12:
@okalachev I'm sorry. I'm testing with the 1.14 beta version of the firmware

okalachev, 2023-06-12:
I guess it might be a PX4 bug, since it a beta version, not Clover's.

youngjae41, 2023-06-13:
@okalachev Thank you for your reply. I will test again when realese comes out.

different qustions ..

In order to use aruco marker, is it true that optical flow must be set to true unconditionally?

youngjae41, 2023-06-13:
PX4 v1.14.0-beta2 testing 

[image]
[image]

Modifying the parameters through airframe works, but gyro and imu don't seem to work. In v1.14.0-beta, the existing iris disappeared and was changed to gazebo_iris_classic, but I don't know what the problem is

okalachev, 2023-06-13:
I would recommend you to switch to stable version of PX4 since I don't see much sense in debugging the beta version for now.

>In order to use aruco marker, is it true that optical flow must be set to true unconditionally?

No, ArUco markers can be used without optical flow.

youngjae41, 2023-06-14:
@okalachev Ok..

In px4 firmware 1.13.3 version and 1.12.3 version, the aircraft fails to take off when operating after setting optical flow in clover.launch to false.

okalachev, 2023-06-19:
There might be problems with PX4's estimator to initialize. Is `force_init` argument `true` in `clover.launch`?

Also try to perform `rosrun clover selfcheck.py` when the drone is not working.

closed 2023-08-05 by youngjae41