CopterExpress/clover #492: only aruco take off error

youngjae41, 2023-08-05:
Hello Long time no see @okalachev 

I am using px4 1.12.3 version firmware. Optical flow is set to false and I am testing using ipython

<arg name="force_init" default="true"/> 
<arg name="simulator" default="true"/> 

set it like this

navigate(x=0, y=0, z=1.0, speed=0.5, frame_id='body', auto_arm=True)

Give the take off command. The gas initially rises and then loses position.

aruco.launch file

    <arg name="aruco_detect" default="true"/>
    <arg name="aruco_map" default="false"/>
    <arg name="aruco_vpe" default="false"/>
    <arg name="placement" default="floor"/> 
    <arg name="length" default="0.22"/> 
    <arg name="map" default="map.txt"/> 
    
    Doesn't it work if I don't use aruco_map?

okalachev, 2023-08-05:
Hi! You can't fly in position mode with optical flow turned off and aruco vpe turned off, since the drone doesn't have the position source.

youngjae41, 2023-08-10:
@okalachev thanks ~! I'm using ipython3 control but land position is strange 
The command is using Land() and using aruco.

okalachev, 2023-08-14:
What do you mean land position is strange? Please, provide more details on that and, if possible, the flight logs.

youngjae41, 2023-08-23:
@okalachev It seems to be a camera problem. Existing clover rpi camera outputs 40 fps, but jetson camera outputs 30 fps max. Could this be why the optical flow isn't working properly?

youngjae41, 2023-08-24:
@okalachev aruco.launch aruco_detect        <param name="use_map_markers" value="true"/> If not set to false, aruco detection will not work.

okalachev, 2023-08-27:
@youngjae41 

>It seems to be a camera problem. Existing clover rpi camera outputs 40 fps, but jetson camera outputs 30 fps max. Could this be why the optical flow isn't working properly?

No, there is no hard requirement to optical flow camera fps to work, 30 FPS is OK.

>aruco.launch aruco_detect If not set to false, aruco detection will not work.

You can use an automated tool that checks your Clover's configuration and prints a full report: `rosrun clover selfcheck.py`.

youngjae41, 2023-08-28:
@okalachev Sorry, misdirection. in the latest version

  <param name="use_map_markers" value="true"/> <- set to false, aruco detection will not work.

okalachev, 2023-08-28:
You can use the above checking tool to diagnose some problems. Also there is log in `/var/log/clover.log` file, if aruco detection falls, there would be errors.

lgh5054, 2023-11-20:
@okalachev Hello, is there a way to land on the Aruco Maker I want?

okalachev, 2023-11-20:
You can use `aruco_X` frame for that, like, `navigate(x=0,y=0,z=0,frame_id='aruco_35')`.

lgh5054, 2023-11-21:
@okalachev Entering Land() RTLs to the place where the aircraft first takek off. Has this ever happened?

okalachev, 2023-11-23:
@lgh5054, can you please reformulate your question? I don't get it.

lgh5054, 2023-11-24:
@okalachev When entering land(), it does not land at the current location, but RTLs to the home point from which it first took off.

okalachev, 2023-11-24:
No, it should not work this way. Maybe it just drifted away for some reason when it was landing?

lgh5054, 2023-11-24:
@okalachev i'm using v 1.14.0 relrease firmware

okalachev, 2023-11-24:
Sorry, this code was not written by us, you can try to contact its author https://t.me/tenessinum to get it.

lgh5054, 2023-12-12:
@okalachev Hello Okalachev Found the problem.

When operating optical flow, even if you hold the drone with your hand and move its position, rviz will initialize it to 0,0,0 coordinates.

optical flow problem or px4 1.14.0 firmware problem?

okalachev, 2025-01-10:
Closed as stale.

okalachev, 2025-01-10:
Closed as stale.

closed 2025-01-10 by okalachev