CopterExpress/clover #495: AttributeError: module 'cv2.aruco' has no attribute 'DetectorParameters_create'

yousef337, 2023-10-11:
I have been following the documentation at 
https://github.com/CopterExpress/clover/blob/master/docs/en/simulation_native.md for running the simulation in ROS Noetic and have faced the following error AttributeError: module 'cv2.aruco' has no attribute 'DetectorParameters_create'.

I am trying to build and run the simulation in an Apptainer writable container based in ROS Noetic (Ubuntu)

okalachev, 2023-10-12:
This is actually strange since the simulation itself doesn't run any Python code that uses OpenCV.

Maybe you're trying to run some OpenCV example using wrong Python version or something like that?

okalachev, 2023-10-14:
OpenCV and `python3-opencv` should be installed during the `rosdep install` step, since `aruco_pose` package [depends](https://github.com/CopterExpress/clover/blob/master/aruco_pose/package.xml#L21C11-L21C20) on `cv_bridge`.

If you somehow don't have these packages installed, then try to install them and repeat.

yousef337, 2023-10-14:
opencv is installed, the problem is not opencv is not there, but aruco_pose called a function in opencv that doesn't exist

okalachev, 2023-10-14:
OK, I guess you have another opencv-python version with changed API. I tried to fix this script for this version, please, pull the repo and try to build again.

closed 2023-10-14 by yousef337