CopterExpress/clover #496: ERROR [px4] Startup script returned with return value: 256

yousef337, 2023-10-11:
I have been following the documentation at
https://github.com/CopterExpress/clover/blob/master/docs/en/simulation_native.md for running the simulation in ROS Noetic and have faced the following error 

Error: no autostart file found (**********/init.d-posix/airframes/4500_clover)
ERROR [px4] Startup script returned with return value: 256

during running roslaunch clover_simulation simulator.launch

okalachev, 2023-10-12:
What virtual machine to you use, and were there any errors during the installation?

Also, can you please provide the full output log of this command, maybe it can help to investigate the reason of this error.

yousef337, 2023-10-12:
I am not using a virtual machine, I am running EndeavourOS that contains an apptainer container running Ubuntu, which I use to run the whole process. The only error that occures was #495, the current error occurs when I rem aruco_pose in order to make the catkin_make run successfully.

okalachev, 2023-10-14:
Are you sure you have symlinked simulated Clover airframe to PX4 sources as required in [this step](https://github.com/CopterExpress/clover/blob/master/docs/en/simulation_native.md#add-the-clover-airframe)?

yousef337, 2023-10-14:
Yes, and checked manually, the file is there

yousef337, 2023-10-14:
After resolving #495, the error stack is the same,

Error: no autostart file found (****/init.d-posix/airframes/4500_clover)
ERROR [px4] Startup script returned with return value: 256
================================================================================REQUIRED process [sitl_0-3] has died!
process has died [pid 558325, exit code 255, cmd ***/devel/lib/px4/px4 ***/src/PX4-Autopilot/ROMFS/px4fmu_common -s etc/init.d-posix/rcS -i 0 __name:=sitl_0 __log:=****/.ros/log/f7aa0c6e-691b-11ee-87cf-84fdd1142c15/sitl_0-3.log].
log file: *****/.ros/log/f7aa0c6e-691b-11ee-87cf-84fdd1142c15/sitl_0-3*.log
Initiating shutdown!
================================================================================
RLException: cannot add process [vpe_publisher-8] after process monitor has been shut down
The traceback for the exception was written to the log file

yousef337, 2023-10-14:
Sometimes, if running the launch file again, gazeboo does open with the empty world, along with the error messages. However, running topics-related command doesn't seem to function properly. 

For example, running rostopic echo /mavros/local_position/pose would result in WARNING: no messages received and simulated time is active.
Is /clock being published?

okalachev, 2023-10-16:
I don't get why px4 binary can't find the `4500_clover` airframe file, if you say, that you have symlinked it as in the manual.

Please, check, if there is a following file: `~/.ros/etc/init.d-posix/airframes/4500_clover`.

yousef337, 2023-11-05:
It is, were you able to replicate the issue? Can you go through the setup docs?

MagicWinnie, 2023-11-22:
Fixed same issue by manually creating a symlink:  
`ln -sf ~/catkin_ws/src/clover/clover_simulation/airframes/*  ~/.ros/etc/init.d-posix/airframes/`  
I guess I have made a mistake while creating a symlink per instructions in docs.

yousef337, 2023-11-22:
Unfortunately it didn't resolve the issue

yousef337, 2023-11-22:
When ls ~/.ros/etc/init.d-posix/airframes/, I would still get 4500_clover. When ls -la, it would yield .ros/etc/init.d-posix/airframes/4500_clover -> src/clover/clover_simulation/airframes/4500_clover

okalachev, 2025-01-10:
Closed as stale.

closed 2025-01-10 by okalachev