CopterExpress/clover #497: Questions regarding ROS2 / ROS1 future plans for adoption

simojo, 2023-10-12:
* Is there any reason why we are committed to using ROS Noetic?
* Are there any plans to migrate this to ROS2 in the future?
* Are there any plans to migrate this to a newer version of ROS1 in the future?

I'm new to the ROS ecosystem, so this question may be coming from an misinformed angle; nevertheless, maybe this is something that, once answered, could be posted to the docs as common knowledge.

okalachev, 2023-10-12:
Hi, thanks for the questions!

1. ROS Noetic is the latest and the current version of ROS 1. I also think, it's the most popular version of ROS at all (both of 1 and 2). I see this version the most frequently in real robotics projects. Most of the Clover's node, I guess, also would work on previous ROS versions, but the main target is ROS Noetic.
2. There is no near plans to migrate this project, as ROS 2 differs dramatically from ROS 1 by its API and tools, and such migration would require to rewrite almost all of the code of the project (not counting rewriting all the documentation). Personally I don't see too many advantages of ROS 2 to do such job, and some solutions of ROS 2 seem actually strange to me. Also, when I checked ROS 2 last time, it lacked a lot of tools and packages that are basically required by this project.
3. ROS Noetic is the newest and the stable version of ROS 1. And ROS team doesn't plan to release any new major versions of ROS 1. This means that the base platform itself (communication and tools) of ROS 1 is stable, but the packages for ROS Noetic may continue to update. 

We get the question about ROS 2 quite often, so yes, maybe this will be added somewhere to the docs.

simojo, 2023-10-12:
Thanks for your gracious reply. I'll close this issue now!

closed 2023-10-12 by simojo