CopterExpress/clover #498: disabling VL53L1X Rangefinder overall

AvishkarSeth, 2023-11-14:
Hi,
Is there any way to disable using VL53L1X Rangefinder as default and only use a vision-based altitude estimation as https://clover.coex.tech/en/aruco_map.html ? I am using Pixhawk 2.4.8 board with Firmware for Pixhawk (Clover 2) – [px4fmu-v2_lpe.px4] version and the RPi image provided with RPi 4.
Thanks!

okalachev, 2023-11-15:
Yes, it's possible. Just set `EKF2_HGT_REF` (or `EKF2_HGT_MODE`) to `Vision` in EKF2, or remove `pub agl as lpos down` from `LPE_FUSION` in LPE, and then the drone will calculate its altitude only from the vision.
We've flown with such a setup a lot, the only problem was the drone couldn't get correction data for its altitude during takeoff stage, but was not a big problem. Other detail is that without the rangefinder, optical flow wouldn't work and should be disabled.

AvishkarSeth, 2023-11-15:
Thanks Okalachev!
I'm having some errors and warnings at the selfcheck command... will try to resolve soon.
It's great to hear the pixhawk 2.4.8 works with the clover framework..
The documentation has been extremely helpful..

okalachev, 2023-11-20:
Can I close the issue now?

AvishkarSeth, 2023-11-20:
Sure. Thank you so much.

closed 2023-11-20 by okalachev