CopterExpress/clover #499: Advice for calling `/gazebo/reset_world` service without ruining calibration

simojo, 2024-02-05:
In the Gazebo simulation environment, when the Clover crashes, I would like to simply call `/gazebo/reset_world` to reset the quadcopter's position. However, this seems to cause a great deal of confusion when `px4_sitl` tries arming again: arming is usually denied due to some kind of miscalibration after calling `/gazebo/reset_world`.

I have also tried the sequence of

1. force-disarming the clover
2. pausing the physics engine
3. resetting its position
4. resuming the physics engine
5. arming the quadcopter,

but this does not work either.

**Are there any suggestions as to how one can reset the Clover's position in gazebo programmatically without causing the simulated sensors to go wild?**

I'm using `clover_vm v1.4` to run my simulations.

okalachev, 2024-02-10:
Hi, sorry for the long time to answer, didn't notice this issue.

As far as I know, the problem with position reset is the architecture of PX4's estimator, EKF2, which doesn't except such jumps to happen. So I just restart the whole simulation when I need to reset. I didn't try to fix that issue yet.

But you may try to restart EKF2 with the following PX4 commands: `ekf2 stop`, `ekf2 start`, maybe it world help? If it does help, I might think how to perform PX4 commands programmatically (MAVLink console is disabled in the simulation, but there may be some solutions to that).

okalachev, 2024-02-28:
Is the issue resolved?

simojo, 2024-02-28:
No, not yet.

okalachev, 2024-04-11:
Have you tried what I advised?

closed 2024-10-04 by okalachev