CopterExpress/clover #506: aruco_pose

Na5050, 2024-06-17:
Есть ли вариант aruco_pose для opencv 4.10?
Дело в том, что микропк у меня orangepi 5  и камеру удалось запустить только с версией opencv 4.10. Но с ней не компилируется оригинальный aruco_pose. Попытался самостоятельно откорректировать исходники, но с aruco_map.cpp возникли проблемы.

okalachev, 2024-06-18:
Здравствуйте, запостите сюда, что за ошибку выдает, полностью.

Na5050, 2024-06-18:
Олег, добрый вечер, в аттаче изменения что я внес, чтобы все скомпилилось, но aruca_map не запускается:
 
[aruco_map-9] restarting process
process[aruco_map-9]: started with pid [5325]
[ INFO] [1718733689.779679249]: ros.nodelet: Loading nodelet /aruco_map of type aruco_pose/aruco_map to manager main_camera_nodelet_manager with the following remappings:
[ INFO] [1718733689.788105617]: ros.nodelet: /camera_info -> /main_camera/camera_info
[ INFO] [1718733689.788168907]: ros.nodelet: /image_raw -> /main_camera/image_raw
[ INFO] [1718733689.788212073]: ros.nodelet: /markers -> /aruco_detect/markers
[Wrn] [ColladaLoader.cc:1928] Triangle input semantic: 'COLOR' is currently not supported
[FATAL] [1718733689.840276124]: ros.nodelet: Failed to load nodelet '/aruco_map` of type `aruco_pose/aruco_map` to manager `main_camera_nodelet_manager'
[Wrn] [ColladaLoader.cc:1928] Triangle input semantic: 'COLOR' is currently not supported
[Wrn] [ColladaLoader.cc:1928] Triangle input semantic: 'COLOR' is currently not supported
[aruco_map-9] process has died [pid 5325, exit code 255, cmd /opt/ros/noetic/lib/nodelet/nodelet load aruco_pose/aruco_map main_camera_nodelet_manager image_raw:=main_camera/image_raw camera_info:=main_camera/camera_info markers:=aruco_detect/markers __name:=aruco_map __log:=/home/orangepi/.ros/log/49e09468-2d9c-11ef-953a-d29751774236/aruco_map-9.log].
log file: /home/orangepi/.ros/log/49e09468-2d9c-11ef-953a-d29751774236/aruco_map-9*.log
 
файла
/home/orangepi/.ros/log/49e09468-2d9c-11ef-953a-d29751774236/aruco_map-9*.log
в логах не обнаружил
 
наверное я не правильно внес изменения в сырцы, может посмотрите? просто с opencv 4.2  у меня камера не работает 

> 
> 
>Здравствуйте, запостите сюда, что за ошибку выдает, полностью.

Na5050, 2024-06-18:
[diff_aruco_detect.patch](https://github.com/user-attachments/files/15891202/diff_aruco_detect.patch)
[diff_aruco_detect.patch](https://github.com/user-attachments/files/15891202/diff_aruco_detect.patch)
[diff_aruco_map.patch](https://github.com/user-attachments/files/15891204/diff_aruco_map.patch)
[diff_aruco_map.patch](https://github.com/user-attachments/files/15891204/diff_aruco_map.patch)
[diff_draw.patch](https://github.com/user-attachments/files/15891205/diff_draw.patch)
[diff_draw.patch](https://github.com/user-attachments/files/15891205/diff_draw.patch)

okalachev, 2024-07-03:
Вообще в ноде aruco_map библиотека для распознавания маркеров vendored, то есть она [скопирована](https://github.com/CopterExpress/clover/tree/master/aruco_pose/vendor/aruco) к нам в исходники. Так что я так и не понял, зачем вам нужны эти патчи.

Na5050, 2024-07-04:
вот это поворот...

видимо, у меня не компилируется эта скопированная библиотека, а используется родная от версии 4.10, и без этих изменений генерируются ошибки такого плана:

/home/orangepi/ws/sim_clov/catkin_ws/src/clover/aruco_pose/src/aruco_map.cpp: In member function ‘void ArucoMap::addMarker(int, double, double, double, double, double, double, double)’:
/home/orangepi/ws/sim_clov/catkin_ws/src/clover/aruco_pose/src/aruco_map.cpp:445:29: error: ‘class cv::aruco::Board’ has no member named ‘dictionary’; did you mean ‘getDictionary’?
  445 |   int num_markers = board_->dictionary->bytesList.rows;
      |                             ^~~~~~~~~~
      |                             getDictionary

...

okalachev, 2024-07-04:
Может у вас получится понять, почему используется не завендоренная версия? Я могу попробовать поразбираться, но это скорее всего будет не скоро.

Na5050, 2024-07-04:
в мейкфайле поменял if ("${OpenCV_VERSION_MINOR}" LESS "9") на if ("${OpenCV_VERSION_MINOR}" LESS "11")
вендор стал компилироваться, но теперь возникли конфликты такого плана

In file included from /home/orangepi/ws/sim_clov/catkin_ws/src/clover/aruco_pose/src/aruco_map.cpp:49:
/home/orangepi/ws/sim_clov/catkin_ws/src/clover/aruco_pose/vendor/aruco/include/opencv2/aruco.hpp:303:20: error: redefinition of ‘class cv::aruco::GridBoard’
  303 | class CV_EXPORTS_W GridBoard : public Board {
      |                    ^~~~~~~~~
In file included from /usr/local/include/opencv4/opencv2/objdetect/aruco_detector.hpp:8,
                 from /usr/local/include/opencv4/opencv2/objdetect.hpp:48,
                 from /usr/local/include/opencv4/opencv2/opencv.hpp:80,
                 from /home/orangepi/ws/sim_clov/catkin_ws/src/clover/aruco_pose/src/aruco_map.cpp:48:
/usr/local/include/opencv4/opencv2/objdetect/aruco_board.hpp:107:27: note: previous definition of ‘class cv::aruco::GridBoard’
  107 | class CV_EXPORTS_W_SIMPLE GridBoard : public Board {
      |                           ^~~~~~~~~

okalachev, 2026-02-07:
stale

closed 2026-02-07 by okalachev