CopterExpress/clover #508: connecting VL53L1X

Na5050, 2024-07-20:
Добрый день, есть намерение использовать px4 1.14.3, в связи с этим возник вопрос: как подключить VL53L1X - к orange pi либо к полетнику

глядя на симулятор, вижу, что он работает без использования VL53L1X, хочу уточнить, действительно ли сам clover не использует на прямую информацию от VL53L1X, и все будет корректно работать, если VL53L1X подключить к полетнику?

и второй вопрос не могу понять, есть ли какие то изменения в прошивке 1.8.2-clover относительно чистой 1.8.2, связанные с получением данных топика  /rangefinder/range, т.е. будет ли получать данные от датчика полетник с прошивкой 1.14.3 если датчик все же подключить к orange pi и будет ли в этом случае все работать корректно?

okalachev, 2024-07-22:
Здравствуйте, подключение к Orange Pi должно быть такое же, как и к Raspberry Pi, то есть к интерфейсу I2C. Драйвер написан, используя API Linux к I2C, так что теоретически должен работать и на Orange Pi, хотя конкретных тестов таких не было.

Симулятор Клевера работает с VL53L1X. Там симулируется дальномер с его свойствами.

Клевер использует информацию напрямую с VL53L1X и передает ее в PX4. Если подключить VL53L1X, то частично сетап работать будет, но не полностью. Например, топик `rangefinder/range` не появится. Для того, чтобы он появился, нужно вносить правки в launch-файлы.

Нет, разницы в получении данных из топика дальномера в разных версиях прошивки никакой нет.

Na5050, 2024-07-23:
спасибо

closed 2024-07-23 by okalachev